CreatorDev/openwrt #11: mmc driver fails if noise on upper bits

Duncan-Frazer-imgtec, 2016-10-21:
Francois Berder: 
Defining all 8 gpios in mmc driver, although only 4 gpios are used, caused some issues during prodtest.
exporting gpio23 (sdhost data6) and configuring it as output confuses the driver and the emmc partition is not displayed anymore in fdisk -l output.
Both emmc and sdcard fails it some gpio are configured as output (gpio23, gpio21)
This can be seen if running the sdcard test then the gpio on GATE.

Duncan-Frazer-imgtec, 2016-10-21:
Hugo:
I am reporting an issue using the Compass click on Ci40.
When booting with the SD card and the Compass click is placed on MikroBUS 1, the boot sequence is interrupted.
Occasionally the same occurs on MikroBUS 2. I've tested the click with a different Ci40 and the issue still persists.

francois-berder-imgtec, 2016-12-09:
This is a hardware bug.
Danube does not respect the datasheet of the sdhost controller which requires that any unused data lines must be pulled high. Hence, mfio21 (MIKROBUS_1INT) mfio22 (MIKROBUS_1_AN), mfio23 (MIKROBUS_1_RST) and mfio24 (MIKROBUS_2_INT) must always be pulled high when using the sdcard.

A possible workaround might be to configure the sdcard driver to use only one data line (obviously slowing down communication with the sdcard). More investigation is needed before applying this change.

closed 2016-12-21 by Ham22