CreatorDev/openwrt #18: when opkg and opkg-smime are both enabled, opkg.conf getting copied from opkg-unsigned and not from opkg-smime

Duncan-Frazer-imgtec, 2016-10-21:
Abhijit Mahajani:
Though we have added a workaround for replace opkg by opkg-smime, we need to fix it properly since we should not change the default openwrt behaviour i.e. DEFAULT_PACKAGES from include/target.mk should not be touched.
Issue is
when opkg and opkg-smime are both enabled, opkg.conf getting copied from opkg-unsigned and not from opkg-smime. so need to fix such that when opkg-smime is enabled it should copy opkg-smime.conf as opkg.conf into target file system.

Ham22, 2016-11-10:
Can we clarify exactly what the requirments for opkg are? Why are we using this pkg when upstream use the normal one. What are the reasons for and against?

abhijit-mahajani-imgtec, 2016-11-24:
1.	Opkg update should work across the different releases. Since it was using a different key build time, opkg update with signature authentication used to fail across the builds. Opkg update used to work fine if board has been booted with the build-all image (i.e. from which build, all packages are generated)
2.	We needed to download the packages from https://downloads.creatordev.io/ 
3.	All packages which are uploaded to this downloads server should be signed via key/pass/certificate.

closed 2016-12-01 by nikhil-zinjurde-imgtec

nikhil-zinjurde-imgtec, 2016-12-02:
Verified that OPKG signing works with newly generated signed packages hosted on a local server.  Packages and image generated from http://jenkins.creatordev.io/job/CreatorDev/job/openwrt/job/ci40/111/

commit 2663b3ee (2016-12-01): Jenkinsfile: add key pair for opkg signing

This connects to #105 and fixes #18

The CI build needs to get key pair from Jenkins credentials manager
and store it in the workspace.