CreatorDev/openwrt #23: cfg80211/mac80211 changes for IMG WiFi support

Duncan-Frazer-imgtec, 2016-10-21:
@abhijit-mahajani-imgtec -
As per https://github.com/openwrt/openwrt/pull/95,
The wifi driver should better use package/kernel/mac80211/ It is only possible to have one mac80211 and cfg80211 subsystem loaded into the kernel at the time and OpenWrt uses mac80211 from backports in package/kernel/mac80211/ . You can either add some patches to package/kernel/mac80211/ or add a separate package like package/kernel/mwlwifi/Makefile

abhijit-mahajani-imgtec, 2016-11-23:
>>You can either add some patches to package/kernel/mac80211/ 
patches have been added at
https://github.com/CreatorDev/openwrt/pull/64/commits/b64e5c22780a14d2a9a0810901d2bb3480a0951b

francois-berder-imgtec, 2016-12-05:
Fixed by:
* https://github.com/CreatorDev/Ci40-platform-feed/pull/102
* https://github.com/CreatorDev/openwrt/pull/64
* https://github.com/CreatorDev/openwrt/pull/83
* https://github.com/CreatorDev/openwrt/pull/81

abhijit-mahajani-imgtec, 2016-12-15:
Closing this one as https://github.com/CreatorDev/openwrt/issues/60 is tracking the upstreaming progress

closed 2016-12-15 by abhijit-mahajani-imgtec