CreatorDev/openwrt #24: WiFi/mac80211.sh restructuring in OpenWrt

Duncan-Frazer-imgtec, 2016-10-21:
@abhijit-mahajani-imgtec -
As per the https://github.com/openwrt/openwrt/pull/95,
there are review comments regarding WiFi/mac80211.sh
1.If we are changing in original mac80211.sh, then just propose a patch with the delta, rather than keeping a copy as target/linux/pistachio/base-files/lib/netifd/wireless/mac80211.sh
2.target/linux/pistachio/base-files/etc/config/wireless should get auto-generated by mac80211.sh at first boot when this devices gets detected.

francois-berder-imgtec, 2016-11-09:
2. target/linux/pistachio/base-files/etc/config/wireless is already generated by mac80211.sh

francois-berder-imgtec, 2016-11-09:
target/linux/pistachio/base-files/lib/netifd/wireless/mac80211.sh is an outdated copy from upstream openwrt (8286b56f4123aa8250d3d35a08dccb4575eaa31f). Once we move to using openwrt mac80211, we can remove target/linux/pistachio/base-files/lib/netifd/wireless/mac80211.sh.

francois-berder-imgtec, 2016-12-05:
Fixed by https://github.com/CreatorDev/openwrt/pull/74.

abhijit-mahajani-imgtec, 2016-12-15:
Closing this one as https://github.com/CreatorDev/openwrt/issues/60 is tracking the upstreaming progress

closed 2016-12-15 by abhijit-mahajani-imgtec