CreatorDev/openwrt #25: Alsa configuration file to be moved to proper package

Duncan-Frazer-imgtec, 2016-10-21:
@abhijit-mahajani-imgtec -
As per https://github.com/openwrt/openwrt/pull/95, regarding target/linux/pistachio/base-files/etc/asound.conf
Doesn't this file belong to the asound package?
Then you can have install script similarly like we have in uboot-env.

manohar-narkhede-imgtec, 2016-12-14:
The Link to mailing list of alsa-lib where the patch is submitted for review:
http://mailman.alsa-project.org/pipermail/alsa-devel/2016-November/115229.html

manohar-narkhede-imgtec, 2017-02-06:
ALSA patch is accepted and applied (http://git.alsa-project.org/?p=alsa-lib.git;a=commit;h=4dfa8f08fb834f7b087f35f9ba1746cd0c989d02).

Shpinkso, 2017-02-06:
@manohar-narkhede-imgtec can this be closed?

manohar-narkhede-imgtec, 2017-02-14:
This issue can be closed. The new issue #292 is created to add patch in LEDE packages till the alsa-lib releases the changes.

closed 2017-02-16 by Ham22