CreatorDev/openwrt #27: Ethernet status LED does not work

Duncan-Frazer-imgtec, 2016-10-21:
@MattAtkinson -
Report from the Creator Forum:
https://forum.creatordev.io/t/green-led-of-ethernet-has-no-activity/109

francois-berder-imgtec, 2016-11-21:
Danube should use the mdio interface to configure the ksz8091 chip but we do not define a mdio node in the device tree and the mdio implementation of the stmmac driver was changed between kernel 4.4 and kernel 4.5.

francois-berder-imgtec, 2016-12-05:
I managed to cherry-pick commits concerning mdio + some fixes concerning micrel/stmac from kernel 4.5 and apply to our fork of linux branch openwrt 4.4.14:

a1ff691 phy: micrel: Disable auto negotiation on startup
17c23af phy: micrel: Ensure interrupts are reenabled on resume
b82eae3 phy: Centralize setting driver module owner
cfffd0a stmmac: Fix 'eth0: No PHY found' regression
73fe42d stmmac: Don't exit mdio registration when mdio subnode is not found in the DTS
f952d4c phy: Add API for {un}registering an mdio device to a bus.
b6ebd49 phy: Add an mdio_device structure
4cdcba9 phy: Centralise print about attached phy
3e93228 phy: phy_{read|write}_mmd_indirect: get addr from phydev
12345e9 phy: add phydev_name() wrapper
3b2dc62 phy: Add phydev_err() and phydev_dbg() macros
a53edf3 phy: Consistently use addr for address on an MII bus
75d16ba mdio: Move allocation of interrupts into core
ad0130d phy: Use phy_read() instead of mdiobus_read()
87eeced mdio: Move mdiobus_read/write operatings into mdio.h
5d256099 stmmac: create of compatible mdio bus for stmmac driver

However, the compat-wireless repository has some preprocessor logic to check if the kernel is below or above version 4.5. So, we would also need to patch this repo.

dks-xueliu, 2017-01-09:
Hello
How is the progress here. Recently I also try to enable ksz8091 phy driver add device tree node to ci40. ksz8091 phy driver can not find the device node but its probe function is still called and this chip can be properly configured. I add some dirty code to enable the LED and it works. So can we keep the current code and try to add a device tree node at first ?

francois-berder-imgtec, 2017-01-09:
No progress so far.
Could you share your code ? I tried adding a device tree node and it did not work.

nikhil-zinjurde-imgtec, 2017-01-16:
@francois-berder-imgtec has this now been fixed by #193?

xueliu, 2017-01-16:
@nikhil-zinjurde-imgtec I believe so.

abhijit-mahajani-imgtec, 2017-01-16:
Yes. Its fixed by https://github.com/CreatorDev/openwrt/pull/193

closed 2017-01-16 by abhijit-mahajani-imgtec