CreatorDev/openwrt #32: TPM reset connected to GPIO

Duncan-Frazer-imgtec, 2016-10-21:
@Ham22 -
Is the fact that the tpm reset line is connected to a Danube gpio a security risk. Need to look into it as not sure but the spec states that you should not be able to re-initialise the chip if the host device is not initialising itself.

Ham22, 2016-10-24:
It seems this is in fact a HW bug and should just have the TPM reset connected to system reset in any future spins.

Ham22, 2016-11-22:
The HW change on future spins will be tracked by HW team. However this ticket is still needed to do work required to make sure we can do the best with what we have right now.

There are two known issues with this:
1) Attacker can reset the tpm whenever they want to reset PCRs (can't be fixed)
2) We ourselves reset the tpm a couple of times during boot when uboot and linux play around with  the GPIO

We should atleast fix number 2 if possible as currently remote attestation won't work due to the PCRs being reset when they shouldn't have been.

Ham22, 2016-12-19:
Further testing is showing that the reset is not working as expected in certain scenarios.

closed 2017-01-03 by Ham22