CreatorDev/openwrt #33: ci40 toolchain default

Duncan-Frazer-imgtec, 2016-10-21:
@Ham22 -
ci40 openwrt uses a lot of defaults that may not necessarily be the best option for us.

- We should discuss which libc to use but glibc is probably better for us as we have strict size limits.
- We should probably configure it to use our codescape sdk toolchain (this would likely be an external prebuilt so would speed up the build massively)

francois-berder-imgtec, 2016-11-04:
I managed to compile an image using an external toolchain  (and use this image) for the Ci40. There was an issue in toolchain/Config.in which did not set some variables.
Since this is not specific to Ci40, I opened  a pull request on upstream openwrt directly at https://github.com/openwrt/openwrt/pull/184.

Ham22, 2016-11-07:
Can you please open this for our repo too as it could be ages before upstream accept it and it would be very useful to have in order to speed up CI builds.

closed 2016-11-25 by Ham22