CreatorDev/openwrt #34: netboot hangs the kernel

Duncan-Frazer-imgtec, 2016-10-21:
@Ham22 -
Netboot is very useful for quick debug cycles when you are making constant changes to the kernel or anything else for that matter.
But it doesn't want to just work so needs some effort put in.

Note saw this patch 0098-pistachio-enable-NFS-client-support.patch in our patch list

francois-berder-imgtec, 2016-11-10:
I could confirm this bug. Here is the error message:

[    2.039832] VFS: Cannot open root device "nfs" or unknown-block(0,255): error -19
[    2.048252] Please append a correct "root=" boot option; here are the available partitions:
[    2.057639] 1f00            1536 mtdblock0  (driver?)
[    2.063351] 1f01               8 mtdblock1  (driver?)
[    2.068986] 1f02               8 mtdblock2  (driver?)
[    2.074679] 1f03             496 mtdblock3  (driver?)
[    2.080363] 1f04          262144 mtdblock4  (driver?)
[    2.085996] 1f05          262144 mtdblock5  (driver?)
[    2.091691] Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(0,255)
[    2.105297] ---[ end Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(0,255)

Ham22, 2016-11-10:
We can get further than that if you apply the patch mentioned above and pass through the correct nfsroot variables from the bootloader. The last issue seems to be that 'tayga' takes over the Ethernet interface which kills everything as it should not be touched when linux is using it for nfs boot.

I've tried disabling it but it seems I don't know enough about networking on openwrt to turn it off completely for Ethernet.

Note also that in our current implementation the Ethernet interface would have an unknown mac address as the scripts currently assign it after boot which could be related however this worked on buildroot.

abhijit-mahajani-imgtec, 2016-11-10:
@Ham22 , It would be better if you already have this somewhere in your fork and point the exact changes, so that not every one needs to do the same changes again. I assume there are some changes needed in config-4.4 and adding kmod-nfsxxx in target.mk?

francois-berder-imgtec, 2016-11-11:
I tried with release 0.9.7 and I found out that I can only use netboot if I use the initramfs-uImage file. I also tried initramfs-uImage from 0.9.4 and it worked.

Compiling the top of the Ci40 branch, copying bin/pistachio/openwrt-pistachio-marduk-marduk_ca8210-initramfs-kernel.itb to /tftpboot/fitImage and use it to boot worked.

Ham22, 2016-11-11:
It was quite a while ago now, looked through my trees and I don't have the patches anymore :(. Sorry. If I get some time I will go through again. But yes it was essentially just adding a kmod for nfs and configuring our bootloader correctly. Also off course removing any scripts/config I could find that were touching eth0 but I didn't finish that bit.

Most of what I did was just taken from the mailing list:
* https://lists.openwrt.org/pipermail/openwrt-devel/2012-February/014213.html

francois-berder-imgtec, 2016-11-14:
For those trying to use netboot, do **not** enable NFS in make kernel_menuconfig. Instead, make menuconfig -> Global build settings, tick the box "Compile the kernel with rootfs on NFS".