CreatorDev/openwrt #36: ci40 kgdb kernel panic

Duncan-Frazer-imgtec, 2016-10-21:
@Ham22 -
As soon as you try to enable kgdb you get a kernel panic with these options enabled:
CONFIG_DEBUG_KERNEL=y
CONFIG_KGDB=y
CONFIG_KGDB_SERIAL_CONSOLE=y
CONFIG_SERIAL_KGDB_NMI=n
CONFIG_KGDB_TESTS=n
CONFIG_KGDB_LOW_LEVEL_TRAP=n
CONFIG_KGDB_KDB=n
root@OpenWrt:/usr/bin# echo ttyS1 > /sys/module/kgdboc/parameters/kgdboc
[61669.361461] CPU 0 Unable to handle kernel paging request at virtual address 36314352, epc == 36314353, ra == 842e4d90
[61669.373481] Oops1:
[61669.376039] CPU: 0 PID: 649 Comm: ash Not tainted 4.1.13 #13
[61669.382359] task: 8fbceaa0 ti: 8f942000 task.ti: 8f942000
[61669.388375] $ 0 : 00000000 00000043 36314353 00000001
[61669.394244] $ 4 : 8ec14664 00000001 00000000 0000002c
[61669.400099] $ 8 : 0a310000 8ed62b86 00000010 00000001
[61669.405954] $12 : 7f983210 00000000 00000000 776fd320
[61669.411848] $16 : 00000000 8ec14664 00000000 8ec14200
[61669.417777] $20 : 84850ee4 00000004 84783d6c 00000020
[61669.423708] $24 : 00000010 84047df8 
[61669.429636] $28 : 8f942000 8f943d48 004631c0 842e4d90
[61669.435573] Hi : 00000000
[61669.438812] Lo : 00000000
[61669.442117] epc : 36314353 0x36314353
[61669.446474] ra : 842e4d90 uart_poll_init+0xfc/0x12c
[61669.452229] Status: 11000403 KERNEL EXL IE 
[61669.456975] Cause : 10800008
[61669.460212] BadVA : 36314352
[61669.463451] PrId : 0001a120 (MIPS interAptiv (multi))
[61669.469203] Modules linked in: pppoe ppp_async iptable_nat pppox ppp_generic pistachio nf_nat_ipv4 nf_conntrack_ipv6 nf_conntrack_ipv4 ipt_REJECT ipt_MASQUERADE xt_time xt_tcpudp xt_state xt_nat xt_multiport xt_mark xt_mac xt_limit xt_id xt_conntrack xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_CT snd_soc_tpa6130a2 slhc sc16is7xx pistachio_internal_dac pistachio_event_timer_module nhc_udp nhc_routing nhc_mobility nhc_ipv6 nhc_hop nhc_fragment nhc_dest nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_nat nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack iptable_raw iptable_mangle iptable_filter ip_tables img_spdif_out img_spdif_in img_parallel_out img_i2s_out img_i2s_in ca8210sm mac802154 ieee802154_6lowpan ieee802154_socket ieee802154 6lowpan snd_soc_core i2c_dev ledtrig_heartbeat ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_raw ip6table_mangle ip6table_filter ip6_tables x_tables uccp420wlan snd_pcm_dmaengine snd_compress snd_pcm_oss snd_mixer_oss snd_pcm snd_timer snd_rawmidi snd_seq_device snd_hwdep snd soundcore dns_resolver leds_gpio
[61669.578077] Process ash (pid: 649, threadinfo=8f942000, task=8fbceaa0, tls=776fd614)
[61669.586728] Stack : 8f5ff330 8f5ff410 847644c0 8ee14100 00000001 00000000 00002580 00000008
0000006e 0000006e 8f40df38 8f3db900 84780000 00000001 8f943dd0 842d56e4
8f058000 840faa4c 00002301 00000001 00000001 84850ee9 00000000 84850000
84780000 84850000 847845ac 8efcc20c 00000030 842eed50 8f943e60 00000000
8ee14100 8f943ef4 00000000 840ee8a4 8f058000 8ed62b80 8f9ec394 84780000
...
[61669.627077] Call Trace:
[61669.630019] [<842d56e4>] tty_find_polling_driver+0x134/0x20c
[61669.636406] [<840faa4c>] dput+0xe0/0x314
[61669.640865] [<842eed50>] configure_kgdboc+0xdc/0x1a0
[61669.646525] [<840ee8a4>] path_cleanup+0x30/0x58
[61669.651740] [<84047e7c>] param_attr_store+0x84/0xbc
[61669.657234] [<840dfa08>] __kmalloc+0x4c/0x13c
[61669.662186] [<84009adc>] mutex_lock+0x14/0x90
[61669.667158] [<84145010>] kernfs_fop_write+0x12c/0x1a8
[61669.672939] [<840e4ad0>] __vfs_write+0x30/0x104
[61669.678101] [<840e7344>] __sb_start_write+0xd4/0x120
[61669.683737] [<840e5160>] rw_verify_area+0xb8/0x150
[61669.689192] [<840e53d0>] vfs_write+0xbc/0x168
[61669.694140] [<84031c08>] SyS_wait4+0xb4/0xe8
[61669.698979] [<84102c50>] __fdget_pos+0x14/0x60
[61669.704019] [<840e5c78>] SyS_write+0x60/0xc4
[61669.708842] [<8402fb3c>] child_wait_callback+0x0/0x68
[61669.714553] [<84002ba8>] syscall_common+0x30/0x54
[61669.719876] 
[61669.721563] 
Code: (Bad address in epc)
[61669.729900] 
[61669.731676] --[ end trace 3ad2abe44f2eea86 ]--

Ham22, 2016-11-17:
@francois-berder-imgtec your comment here https://forum.creatordev.io/t/query-debug-kernel-with-kgdb-enable/152/2 suggests this is no longer an issue?

francois-berder-imgtec, 2016-11-17:
I added a reply which contains some instructions on how to use kgdb on the Ci40.

closed 2016-11-17 by francois-berder-imgtec