CreatorDev/openwrt #37: Reboot command sometimes causes eth0 to cycle between up & down

Duncan-Frazer-imgtec, 2016-10-21:
@mtusnio -
Randomly, when rebooting the board with reboot, the networking settings take about 160s to load. During that time the system makes multiple attempts at this:
[ 21.847111] img/uccp420wlan/MCP_LOADER.ldr is loaded
[ 21.923710] img/uccp420wlan/MAC_LOADER.ldr is loaded
Finally it brings up eth0 and wlan only, however it causes the eth0 port to constantly switch between up & down states if an ethernet cable is plugged in.

[ 1765.399616] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[ 1767.380441] stmmaceth 18140000.ethernet eth0: Link is Up - 100Mbps/Full - flo
w control rx/tx
[ 1767.380598] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[ 1767.618905] of_get_named_gpiod_flags: parsed 'gpios' property of node '/gpio_
leds/user1[0]' - status (-517)
[ 1767.637553] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready

Duncan-Frazer-imgtec, 2016-10-21:
Needs to be assigned to Yi Laing

yi-liang, 2016-11-08:
adding error info in the log, 
but in this case, unplug the eth cable + power reset the board can stop the problem (we thought only flash the board again can fix it)

[   62.226397] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[   64.210323] stmmaceth 18140000.ethernet eth0: Link is Up - 10Mbps/Full - flow control rx/tx
[   64.219862] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   64.475520] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[   66.044632] ca8210 spi0.0: ca8210 was busy during attempted write
[   66.070185] ca8210 spi0.0: spi write retry 1...
[   66.090497] ca8210 spi0.0: ca8210 was busy during attempted write
[   66.110129] ca8210 spi0.0: spi write retry 1...
[   66.130299] ca8210 spi0.0: spi read failed, returned -50
[   66.136735] ca8210 spi0.0: ca8210 was busy during attempted write
[   66.160140] ca8210 spi0.0: spi write retry 1...
[   66.450360] stmmaceth 18140000.ethernet eth0: Link is Up - 10Mbps/Full - flow control rx/tx
[   66.459897] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   66.714961] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[   67.016167] ca8210 spi0.0: ca8210 was busy during attempted write
[   67.040153] ca8210 spi0.0: spi write retry 1...
[   68.050139] ca8210 spi0.0: Synchronous confirm timeout
[   68.055891] ca8210 spi0.0: error setting max be, MLME-SET.confirm status = 255
[   68.690276] stmmaceth 18140000.ethernet eth0: Link is Up - 10Mbps/Full - flow control rx/tx
[   68.699819] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   68.926325] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[   70.910309] stmmaceth 18140000.ethernet eth0: Link is Up - 10Mbps/Full - flow control rx/tx
[   70.919770] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready

mtusnio, 2016-11-21:
From my experience usually this is preceded by a failure to get a lock file set up when running fw_printenv. If you reboot the board to see if that fixes the problem with getting variables from the bootloader I've noticed that every time I get the eth0 link cycling error after the subsequent boot.

Ham22, 2016-11-29:
@nikhil-zinjurde-imgtec can we run a soak that confirms if this is the cascoda driver. Just remove the lowpan/cascoda init scripts so that the module is never loaded.

nikhil-zinjurde-imgtec, 2016-12-01:
@Ham22 The soak test has not shown the up-down cycle issue, even on the TI board, after removing cascoda kmod package and the lowpan init script.

Shpinkso, 2016-12-02:
Aim of this ticket now is to retest with latest cascoda driver and ensure we have some more detailed information on where the issue is caused.
If we're confident it is cascoda driver related, then raise a ticket with them.

nikhil-zinjurde-imgtec, 2016-12-06:
Preliminary tests show that with the latest version of OpenWrt v0.10.6 and U-boot v1.0.2, the problem is no longer seen on both ca8210 and cc2520 based boards.

nikhil-zinjurde-imgtec, 2016-12-08:
Repeated tests show that this issue is no longer seen or reproducible with the CreatorDev/OpenWrt v0.10.6 and CreatorDev/u-boot v1.0.2.  So, closing this now

closed 2016-12-08 by nikhil-zinjurde-imgtec