CreatorDev/openwrt #40: UDP bandwidth over Ethernet issues

Duncan-Frazer-imgtec, 2016-10-21:
@mtusnio -
Copying files using scp/wget over Ethernet seems to max up the bandwith at about 2.2 MB/s. Additionally, running iperf3 over UDP nets the following results:

**Local connection (CI40 to PC)**
iperf3 -c 192.168.0.1 -i 1 -u -b 200M
Connecting to host 192.168.0.1, port 5201
[ 4] local 192.168.0.2 port 49873 connected to 192.168.0.1 port 5201
[ ID] Interval Transfer Bandwidth Total Datagrams
[ 4] 0.00-1.00 sec 1.90 MBytes 15.9 Mbits/sec 243 
[ 4] 1.00-2.00 sec 2.02 MBytes 17.0 Mbits/sec 259 
[ 4] 2.00-3.00 sec 2.08 MBytes 17.4 Mbits/sec 266 
[ 4] 3.00-4.00 sec 2.09 MBytes 17.5 Mbits/sec 267 
[ 4] 4.00-5.00 sec 2.08 MBytes 17.4 Mbits/sec 266 
[ 4] 5.00-6.00 sec 2.09 MBytes 17.5 Mbits/sec 267 
[ 4] 6.00-7.00 sec 2.08 MBytes 17.4 Mbits/sec 266 
[ 4] 7.00-8.00 sec 2.09 MBytes 17.5 Mbits/sec 267 
[ 4] 8.00-9.00 sec 2.09 MBytes 17.5 Mbits/sec 267 
[ 4] 9.00-10.00 sec 2.09 MBytes 17.5 Mbits/sec 267 
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval Transfer Bandwidth Jitter Lost/Total Datagrams
[ 4] 0.00-10.00 sec 20.6 MBytes 17.3 Mbits/sec 0.762 ms 10/2634 (0.38%) 
[ 4] Sent 2634 datagrams
iperf Done.

**Iperf3 server**
iperf3 -c bouygues.testdebit.info -i 1 -u -b 200M
Connecting to host bouygues.testdebit.info, port 5201
[ 4] local 10.40.24.6 port 60806 connected to 89.84.127.53 port 5201
[ ID] Interval Transfer Bandwidth Total Datagrams
[ 4] 0.00-1.00 sec 1.86 MBytes 15.6 Mbits/sec 238 
[ 4] 1.00-2.00 sec 2.12 MBytes 17.8 Mbits/sec 271 
[ 4] 2.00-3.00 sec 2.14 MBytes 18.0 Mbits/sec 274 
[ 4] 3.00-4.00 sec 2.09 MBytes 17.5 Mbits/sec 267 
[ 4] 4.00-5.00 sec 2.10 MBytes 17.6 Mbits/sec 269 
[ 4] 5.00-6.00 sec 2.13 MBytes 17.9 Mbits/sec 273 
[ 4] 6.00-7.00 sec 2.09 MBytes 17.6 Mbits/sec 268 
[ 4] 7.00-8.00 sec 2.09 MBytes 17.5 Mbits/sec 267 
[ 4] 8.00-9.00 sec 2.07 MBytes 17.4 Mbits/sec 265 
[ 4] 9.00-10.00 sec 2.09 MBytes 17.6 Mbits/sec 268 
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval Transfer Bandwidth Jitter Lost/Total Datagrams
[ 4] 0.00-10.00 sec 20.8 MBytes 17.4 Mbits/sec 0.775 ms 0/2660 (0%) 
[ 4] Sent 2660 datagrams
iperf Done.

Duncan-Frazer-imgtec, 2016-10-21:
@MattAtkinson -
**This problem seems to only affect UDP transmit. Receive works fine.**
In iperf3, I get the results shown above with the following command (Client on Ci40, server on laptop):
iperf3 -c <IP> -u -b 200M
but if I do:
iperf3 -c <IP> - u -b -200M -w 150k
I get really good results (over 90mbits). It seems like setting the window size between 130k and 150k starts to improve the speed dramatically. I don't understand why this would be the case, but maybe something needs to be looked at?

manohar-narkhede-imgtec, 2017-02-24:
I tested the latest source code with above iperf3 commands and it seems the data is getting transmitted from Ci40 to Linux machine at maximum 380Mbps. 
-------------------------------------------------------------------------------------------------------------------
root@OpenWrt:/# iperf3 -c 192.168.91.82 -u -b 500M
Connecting to host 192.168.91.82, port 5201
[  4] local 192.168.91.154 port 36639 connected to 192.168.91.82 port 5201
[ ID] Interval           Transfer     Bandwidth       Total Datagrams
[  4]   0.00-1.00   sec  40.8 MBytes   342 Mbits/sec  5216
[  4]   1.00-2.00   sec  45.1 MBytes   378 Mbits/sec  5774
[  4]   2.00-3.00   sec  44.9 MBytes   377 Mbits/sec  5746
[  4]   3.00-4.00   sec  45.1 MBytes   379 Mbits/sec  5778
[  4]   4.00-5.00   sec  45.0 MBytes   378 Mbits/sec  5764
[  4]   5.00-6.00   sec  45.6 MBytes   382 Mbits/sec  5834
[  4]   6.00-7.00   sec  46.0 MBytes   386 Mbits/sec  5884
[  4]   7.00-8.00   sec  46.0 MBytes   386 Mbits/sec  5884
[  4]   8.00-9.00   sec  45.9 MBytes   385 Mbits/sec  5871
[  4]   9.00-10.00  sec  45.8 MBytes   384 Mbits/sec  5857
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bandwidth       Jitter    Lost/Total Datagrams
[  4]   0.00-10.00  sec   450 MBytes   378 Mbits/sec  0.751 ms  44730/57413 (78%)
[  4] Sent 57413 datagrams
-------------------------------------------------------------------------------------------------------------------
SCP is very simple tool to simple copy files back and forth. It was not designed to super-fast speeds and it has really small buffers on both sides.

The different tools like nc can be used for fast transfer(http://intermediatesql.com/linux/scrap-the-scp-how-to-copy-data-fast-using-pigz-and-nc/).

Ham22, 2017-02-24:
The comments above suggest this is nothing to do with network issues but actually is just a limitation of the embedded system. Closing, please re-open if you feel there is an issue that needs fixing.

closed 2017-02-24 by Ham22