CreatorDev/openwrt #41: Get BTLE working in OpenWrt

Duncan-Frazer-imgtec, 2016-10-21:
@abhijit-mahajani-imgtec

tjobanpu, 2016-11-09:
The scanning of the BLEE using the command
hcitool lescan
oot@OpenWrt:~# hcitool lescan                                                  
LE Scan ...                                                                     
41:68:C8:74:C7:44 (unknown)                                                     
41:68:C8:74:C7:44 (unknown)                                                     
BC:6A:29:AB:E1:48 (unknown)                                                     
BC:6A:29:AB:E1:48 SensorTag                                                     
41:68:C8:74:C7:44 (unknown)                                                     
FD:A7:0B:36:1D:DD (unknown)                                                     
FD:A7:0B:36:1D:DD Charge HR                                                     
78:A5:04:8C:23:C5 (unknown)                                                     
78:A5:04:8C:23:C5 SensorTag                                                     
41:68:C8:74:C7:44 (unknown)                                                     
FD:A7:0B:36:1D:DD (unknown)                                                     
FD:A7:0B:36:1D:DD Charge HR                                                     
43:C1:9D:EC:AC:4B (unknown)    

hcitool scan
root@OpenWrt:~# hcitool scan                                                  
Scanning ...                                                                    
        88:53:2E:AF:38:7B       n/a                                             
        E2:2F:F2:1E:A6:FA       CSR - bc7                                       
        E2:27:7A:56:ED:E6       n/a                                             
        E2:21:0A:55:85:4F       n/a.- bc7                                       
        E2:2F:3C:B5:F7:D9       CSR - bc7                                       
        E2:26:BF:2B:BF:71       CSR - bc7                                       
        E2:26:F6:DE:9E:8F       CSR - bc7                                       
        E2:2C:46:52:CD:F0       CSR - bc7                                       
        00:19:F5:87:96:36       CSR - bc7                                       
        E2:2C:E6:1A:B7:DC       CSR - bc7.                                      
        E2:21:FB:D7:5B:AB       CSR - bc7                                       
        E2:2C:3C:A3:66:DB       CSR - bc7                                       
        E2:28:87:2C:4B:65       CSR - bc7                                       
        E2:2A:38:E6:95:4F       CSR - bc7                                       
        E2:25:B7:0C:82:75       CSR - bc7                                       
        00:19:F5:86:0D:64       Jongo T4 8602EE.

francois-berder-imgtec, 2016-11-28:
This task should also enable by default the bluez-utils package.

closed 2017-01-18 by Ham22