CreatorDev/openwrt #42: Enable bluez package

Duncan-Frazer-imgtec, 2016-10-21:
@francois-berder-imgtec -
We need hci, hciconfig and hcitool to run test_bluetooth and test_ble
gatttool cannot be added because of license incompatibility.

bluez depends on glib2 which depends on iconv.
The current build of glib2 is broken and a pull request exists to fix the issue: https://github.com/openwrt/packages/pull/2425. However, this pull request is now closed.
Delaying this task until build of glib2 is fixed.

It seems that glib2 is compiling for host then for target.
iconv-full does not have host configuration in feed. But iconv is compiled for host. Hence, during the configuration of glib2 it uses header and libraries from iconv instead of iconv-full.

See my comments here: https://github.com/openwrt/packages/pull/2322.

francois-berder-imgtec, 2016-10-25:
bluez package relies on readline library which has GPLv3 license. I do not know if it means we cannot enable it by default. Anyway, the bluez package now  builds.

francois-berder-imgtec, 2016-11-28:
Closing this issue. It has been squashed into  #41

closed 2016-11-28 by francois-berder-imgtec