CreatorDev/openwrt #44: Add documentation regarding Wi-Fi and 6LoWPAN interference

Duncan-Frazer-imgtec, 2016-10-21:
@nikhil-zinjurde-imgtec -
We should make developers of Ci40 and Creator Kit aware of the reasons for having Wi-Fi and 6LoWPAN on channels far away from each other. We should also mention the default channel (i.e. 26) being used for our demo applications

Duncan-Frazer-imgtec, 2016-10-21:
@MattAtkinson -
That sounds like an FAQ type thing. If someone knows enough to start changing the 6LoWPAN channels (we don't explain how to do this) then they'll know about interference with standard WiFi. Definitely not for the Creator Kit "getting started" document. We'll need some sort of FAQ/knowledgebase for information like this - I'll ask around to see what we're planning for this.

nikhil-zinjurde-imgtec, 2016-12-02:
I have raised this as a post on Creator Forum, so this issue can be even closed, as there is nothing to be done in openwrt github project

closed 2016-12-06 by abhijit-mahajani-imgtec