CreatorDev/openwrt #45: Show stages of boot with LED feedback

Duncan-Frazer-imgtec, 2016-10-21:
It would be useful for headless demos if the user could tell which stage of the boot sequence the device is at by some sort of LED change. For example, all LEDs might be turned on by the bootloader, then one LED might turn off when the kernel is loaded, another for init started, another once LWM2M daemons are running.
At the minimum, a single LED that indicates the device is ready would be really useful, as the demoer can wait for the LED before proceeding.

Ham22, 2016-10-24:
u-boot has some of this built in which  we can simply turn on. Note however the least cost would be to just accept the pre-defined sequences already there. It also only covers up to the point linux takes over, from then on would need something else.

See https://github.com/CreatorDev/u-boot/blob/2015.10/README#L2284

francois-berder-imgtec, 2016-11-30:
This pr https://github.com/CreatorDev/u-boot/pull/20 fixes this issue.

closed 2016-12-02 by Ham22