CreatorDev/openwrt #50: Wifi AP startup error when using non default config

Duncan-Frazer-imgtec, 2016-10-24:
@abhijit-mahajani-imgtec 
prof on the creator forum confirmed that the startup issues he's having affect both 0.9.4 and 0.10.4:
Paste from https://forum.creatordev.io/t/wifi-ap-radio-settings-broken/210:

I've got a problem with the WiFi AP configuration, as soon as I make any change to the radio configuration from the defaults WiFI doesn't ever recover and I'll only receive those messages in the system log:

[ 422.867171] img/uccp420wlan/MCP_LOADER.ldr is loaded
[ 422.946554] img/uccp420wlan/MAC_LOADER.ldr is loaded
[ 424.524722] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[ 434.265328] img/uccp420wlan/MCP_LOADER.ldr is loaded
[ 434.348674] img/uccp420wlan/MAC_LOADER.ldr is loaded
[ 435.944295] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[ 447.741800] img/uccp420wlan/MCP_LOADER.ldr is loaded
[ 447.821123] img/uccp420wlan/MAC_LOADER.ldr is loaded
[ 449.404827] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
To reproduce I do a factory reset, then create an AP with the default settings, verify that the AP works, then switch the Operating Mode from "Legacy" to "AC" and from that point on (even when switching back to "Legacy") the AP doesn't work anymore. The webinterface shows "Wireless is disabled or not associated" as status.

tjobanpu, 2016-11-14:
The below issues were noted.
ACS was not implemented in the wifi driver and it was the default mode in the configuration. What I mean is when you switch from the legacy mode to AC mode the hostapd configuration file has the channel=acs_survey by default.

changed this channel number to 36 manually and removed the HT and VHT bandwidth of 40 and 80 Mhz .

The other issue was hostapd does not work in 5 Ghz with the reg domain set as global! numberic value of the country code set to 0.
country_code=00

Removed the above line and set the country code manually from the below command.

iw reg set GB

the above requires us to manually remove the flags ieee80211h and ieee80211d from the hosttapd config file located in "/var/run/hostapd-phy0.conf "

The last issue I faced was the compilation did not include the ieee80211n and the hostapd was constantly complaining about it, I had to manually set the corresponding variable in the Makefile as it was not obeying the .config file in which I had set the variable. This could be my build environment issue but need to confirm.

tjobanpu, 2016-11-30:
Raised a PR 
https://github.com/CreatorDev/uccp420wlan/pull/6

Ham22, 2016-11-30:
@tjobanpu waffle expects connections from the PR to here for linking to work. Please remove the comment here and instead add a 'connects to CreatorDev/openwrt#50' in the description (not as a comment) of the PR that fixes it.

tjobanpu, 2016-12-14:
The 11ac is working by changing the country code this needs to be documented to the user

abhijit-mahajani-imgtec, 2017-01-03:
Documentation needs to be updated

closed 2017-01-03 by abhijit-mahajani-imgtec