CreatorDev/openwrt #65: 'syntax error: invalid arithmetic operator' when updating feeds

MattAtkinson, 2016-11-14:
From current tip OpenWrt, if you run "./scripts/feeds update -a" you get lots of messages about syntax errors. It seems like everything works correctly, but it looks like it's broken.

```Updating feed 'packages' from 'https://github.com/openwrt/packages.git' ...
Cloning into './feeds/packages'...
remote: Counting objects: 3722, done.
remote: Compressing objects: 100% (3132/3132), done.
remote: Total 3722 (delta 143), reused 3007 (delta 91), pack-reused 0
Receiving objects: 100% (3722/3722), 2.29 MiB | 838.00 KiB/s, done.
Resolving deltas: 100% (143/143), done.
Checking connectivity... done.
Create index file './feeds/packages.index' 
/home/img/creator/toolchain/openwrt/scripts/getver.sh: line 22: v0.10.5: syntax error: invalid arithmetic operator (error token is ".10.5")
/home/img/creator/toolchain/openwrt/scripts/getver.sh: line 22: v0.10.5: syntax error: invalid arithmetic operator (error token is ".10.5")
/home/img/creator/toolchain/openwrt/scripts/getver.sh: line 22: v0.10.5: syntax error: invalid arithmetic operator (error token is ".10.5")
Collecting package info: done
/home/img/creator/toolchain/openwrt/scripts/getver.sh: line 22: v0.10.5: syntax error: invalid arithmetic operator (error token is ".10.5")
Collecting target info: done```

abhijit-mahajani-imgtec, 2016-11-15:
@MattAtkinson , how are you updating feeds?
`Updating feed 'packages' from 'https://github.com/openwrt/packages.git' ...
Cloning into './feeds/packages'...
remote: Counting objects: 3722, done.
remote: Compressing objects: 100% (3132/3132), done.
remote: Total 3722 (delta 143), reused 3007 (delta 91), pack-reused 0
Receiving objects: 100% (3722/3722), 2.29 MiB | 837.00 KiB/s, done.
Resolving deltas: 100% (143/143), done.
Checking connectivity... done.
Create index file './feeds/packages.index'
Collecting package info: done
Collecting target info: done
Updating feed 'luci' from 'https://github.com/openwrt/luci.git' ...
Cloning into './feeds/luci'...
remote: Counting objects: 4241, done.
remote: Compressing objects: 100% (2252/2252), done.
remote: Total 4241 (delta 998), reused 2977 (delta 530), pack-reused 0
Receiving objects: 100% (4241/4241), 3.65 MiB | 524.00 KiB/s, done.
Resolving deltas: 100% (998/998), done.
Checking connectivity... done.
Create index file './feeds/luci.index'
Collecting package info: done
Collecting target info: done
Updating feed 'routing' from 'https://github.com/openwrt-routing/packages.git' ...
Cloning into './feeds/routing'...
remote: Counting objects: 351, done.
remote: Compressing objects: 100% (278/278), done.
remote: Total 351 (delta 25), reused 253 (delta 14), pack-reused 0
Receiving objects: 100% (351/351), 229.69 KiB | 233.00 KiB/s, done.
Resolving deltas: 100% (25/25), done.
Checking connectivity... done.
Create index file './feeds/routing.index'
Collecting package info: done
Collecting target info: done
Updating feed 'telephony' from 'https://github.com/openwrt/telephony.git' ...
Cloning into './feeds/telephony'...
remote: Counting objects: 178, done.
remote: Compressing objects: 100% (160/160), done.
remote: Total 178 (delta 11), reused 111 (delta 5), pack-reused 0
Receiving objects: 100% (178/178), 109.90 KiB | 0 bytes/s, done.
Resolving deltas: 100% (11/11), done.
Checking connectivity... done.
Create index file './feeds/telephony.index'
Collecting package info: done
Collecting target info: done
Updating feed 'management' from 'https://github.com/openwrt-management/packages.git' ...
Cloning into './feeds/management'...
remote: Counting objects: 53, done.
remote: Compressing objects: 100% (41/41), done.
remote: Total 53 (delta 9), reused 35 (delta 1), pack-reused 0
Unpacking objects: 100% (53/53), done.
Checking connectivity... done.
Create index file './feeds/management.index'
Collecting package info: done
Collecting target info: done
Updating feed 'targets' from 'https://github.com/openwrt/targets.git' ...
Cloning into './feeds/targets'...
remote: Counting objects: 159, done.
remote: Compressing objects: 100% (107/107), done.
remote: Total 159 (delta 28), reused 141 (delta 25), pack-reused 0
Receiving objects: 100% (159/159), 149.24 KiB | 153.00 KiB/s, done.
Resolving deltas: 100% (28/28), done.
Checking connectivity... done.
Create index file './feeds/targets.index'
Collecting package info: done
Collecting target info: done
Updating feed 'ci40' from 'https://github.com/CreatorDev/Ci40-platform-feed.git' ...
Cloning into './feeds/ci40'...
remote: Counting objects: 40, done.
remote: Compressing objects: 100% (35/35), done.
remote: Total 40 (delta 1), reused 31 (delta 1), pack-reused 0
Unpacking objects: 100% (40/40), done.
Checking connectivity... done.
Create index file './feeds/ci40.index'
Collecting package info: done
Collecting target info: done
`

Even installing feeds also worked for me.
`./scripts/feeds install -a
Installing all packages from feed packages.
Installing all packages from feed luci.
Installing all packages from feed routing.
Installing all packages from feed telephony.
Installing all packages from feed management.
Installing all packages from feed targets.
Installing all packages from feed ci40.
`

MattAtkinson, 2016-11-15:
@abhijit-mahajani-imgtec 
all I did was git clone openwrt and then run './scripts/feeds update -a'. Just tried it again and got the same result. I'm using Ubuntu 16.04. The message was slightly different this time as the "error token" was .10.5-5-gfd72bb4

Ham22, 2016-11-15:
This is a know issue right? You just need to move our top level getver.sh script to scripts directory to replace the one already in there.

abhijit-mahajani-imgtec, 2016-11-17:
It has been fixed in #67

closed 2016-11-17 by abhijit-mahajani-imgtec