CreatorDev/openwrt #71: Build fails due to missing file on upstream downloads server

MattAtkinson, 2016-11-15:
Current tip OpenWrt does not build as the requested ca-certificates file cannot be found on the server. See logs:

```make[3]: Entering directory '/home/img/creator/toolchain/nopr/openwrt/package/system/ca-certificates'
mkdir -p /home/img/creator/toolchain/nopr/openwrt/dl
SHELL= flock /home/img/creator/toolchain/nopr/openwrt/tmp/.ca-certificates_20160104.tar.xz.flock -c '	/home/img/creator/toolchain/nopr/openwrt/scripts/download.pl "/home/img/creator/toolchain/nopr/openwrt/dl" "ca-certificates_20160104.tar.xz" "d9665a83d0d3ef8176a38e6aa20458e9" "" "http://ftp.debian.org/debian/pool/main/c/ca-certificates"'
--2016-11-15 14:48:14--  http://ftp.debian.org/debian/pool/main/c/ca-certificates/ca-certificates_20160104.tar.xz
Resolving ftp.debian.org (ftp.debian.org)... 130.89.148.12, 2001:610:1908:b000::148:12
Connecting to ftp.debian.org (ftp.debian.org)|130.89.148.12|:80... connected.
HTTP request sent, awaiting response... 404 Not Found
2016-11-15 14:48:14 ERROR 404: Not Found.

Download failed.
--2016-11-15 14:48:14--  http://mirror2.openwrt.org/sources/ca-certificates_20160104.tar.xz
Resolving mirror2.openwrt.org (mirror2.openwrt.org)... 46.4.11.11
Connecting to mirror2.openwrt.org (mirror2.openwrt.org)|46.4.11.11|:80... connected.
HTTP request sent, awaiting response... 404 Not Found
2016-11-15 14:48:14 ERROR 404: Not Found.

Download failed.
--2016-11-15 14:48:14--  http://downloads.openwrt.org/sources/ca-certificates_20160104.tar.xz
Resolving downloads.openwrt.org (downloads.openwrt.org)... 78.24.191.177
Connecting to downloads.openwrt.org (downloads.openwrt.org)|78.24.191.177|:80... connected.
HTTP request sent, awaiting response... 404 Not Found
2016-11-15 14:48:14 ERROR 404: Not Found.

Download failed.
No more mirrors to try - giving up.
Makefile:61: recipe for target '/home/img/creator/toolchain/nopr/openwrt/dl/ca-certificates_20160104.tar.xz' failed
make[3]: *** [/home/img/creator/toolchain/nopr/openwrt/dl/ca-certificates_20160104.tar.xz] Error 2
make[3]: Leaving directory '/home/img/creator/toolchain/nopr/openwrt/package/system/ca-certificates'
package/Makefile:196: recipe for target 'package/system/ca-certificates/compile' failed
make[2]: *** [package/system/ca-certificates/compile] Error 2
make[2]: Leaving directory '/home/img/creator/toolchain/nopr/openwrt'
package/Makefile:193: recipe for target '/home/img/creator/toolchain/nopr/openwrt/staging_dir/target-mipsel_mips32_musl-1.1.15/stamp/.package_compile' failed
make[1]: *** [/home/img/creator/toolchain/nopr/openwrt/staging_dir/target-mipsel_mips32_musl-1.1.15/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/img/creator/toolchain/nopr/openwrt'
/home/img/creator/toolchain/nopr/openwrt/include/toplevel.mk:192: recipe for target 'world' failed
make: *** [world] Error 2
```

Ham22, 2016-11-16:
Updating to match the latest code should fix this now but we should keep this ticket open to track a permanent solution which would be to add our own download mirror with a cache of all the packages we currently use.

Ham22, 2016-11-19:
Looking at the code this mirror already exists and is already configured in our defconfigs.

The issue is that it's only used if CONFIG_DEVEL is also enabled. I suggest we remove from the config file and have the CI system add both required configs options as part of the build.

closed 2016-11-23 by Ham22