CreatorDev/openwrt #84: Add the bounce buffers patch for WiFi

Shpinkso, 2016-11-21:


francois-berder-imgtec, 2016-11-21:
Currently, adding the patch which enables GENERIC_ISA_DMA causes 6lowpan and wifi driver crash:

[    8.704090] Starting module ca8210
[    8.708163] ca8210 spi0.0: Inserting ca8210
[    8.734965] kmodloader: page allocation failure: order:0, mode:0x26000c1
[    8.742611] CPU: 3 PID: 761 Comm: kmodloader Not tainted 4.4.14 #1
[    8.749501] Stack : 00000000 00000000 8477688a 00000036 00000000 00000000 84700000 84700000
          8f18db60 846fc743 8460f3b4 00000003 000002f9 84773650 00000000 024000c0
          00000040 840705ac 00000040 840703e4 00000000 00000004 84615174 8f3099ac
          00000000 8406e348 00000000 00000000 8477688a 84700000 00000000 003099ac
          00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
          ...
[    8.789238] Call Trace:
[    8.792000] [<84013c50>] show_stack+0x50/0x84
[    8.796908] [<84263a78>] dump_stack+0xd4/0x10c
[    8.801896] [<840afff0>] warn_alloc_failed+0xe8/0x10c
[    8.807536] [<840b2820>] __alloc_pages_nodemask+0x5a8/0x718
[    8.813775] [<840e5648>] allocate_slab+0xcc/0x2b4
[    8.819025] [<840e731c>] ___slab_alloc.isra.9.constprop.17+0x250/0x324
[    8.826306] [<840e7410>] __slab_alloc.isra.10.constprop.16+0x20/0x44
[    8.833387] [<840e757c>] __kmalloc+0x148/0x234
[    8.838409] [<8f3f24e8>] init_module+0x3164e8/0x316e40 [ca8210sm]
[    8.845231] 
[    8.846973] Mem-Info:
[    8.849544] active_anon:97 inactive_anon:2 isolated_anon:0
[    8.849544]  active_file:309 inactive_file:217 isolated_file:0
[    8.849544]  unevictable:0 dirty:1 writeback:0 unstable:0
[    8.849544]  slab_reclaimable:403 slab_unreclaimable:1290
[    8.849544]  mapped:340 shmem:2 pagetables:33 bounce:0
[    8.849544]  free:43374 free_pcp:2 free_cma:0
[    8.884765] DMA free:0kB min:0kB low:0kB high:0kB active_anon:0kB inactive_anon:0kB active_file:0kB inactive_file:0kB unevictable:0kB isolated(anon):0kB isolated(file):0kB present:65536kB managed:0kB s
[    8.929097] lowmem_reserve[]: 0 180 180
[    8.933486] Normal free:173548kB min:1720kB low:2148kB high:2580kB active_anon:348kB inactive_anon:8kB active_file:1240kB inactive_file:828kB unevictable:0kB isolated(anon):0kB isolated(file):0kB preso
[    8.981876] lowmem_reserve[]: 0 0 0
[    8.985781] DMA: 0*4kB 0*8kB 0*16kB 0*32kB 0*64kB 0*128kB 0*256kB 0*512kB 0*1024kB 0*2048kB 0*4096kB = 0kB
[    8.996646] Normal: 103*4kB (UME) 58*8kB (UME) 22*16kB (UM) 14*32kB (UME) 4*64kB (M) 5*128kB (M) 2*256kB (ME) 3*512kB (M) 1*1024kB (M) 2*2048kB (ME) 40*4096kB (M) = 173580kB
[    9.014101] 527 total pagecache pages
[    9.018189] 0 pages in swap cache
[    9.021926] Swap cache stats: add 0, delete 0, find 0/0
[    9.027744] Free swap  = 0kB
[    9.030994] Total swap = 0kB
[    9.034203] 65536 pages RAM
[    9.037311] 0 pages HighMem/MovableOnly
[    9.041624] 19237 pages reserved
[    9.045242] ca8210 spi0.0: ca8210_dev_com_init failed
[    9.160198] ca8210 spi0.0: Removing ca8210
[    9.164794] ca8210 spi0.0: Switching external clock off
[    9.171021] ca8210 spi0.0: Bufferless transfer has length 2
[    9.177299] CPU 3 Unable to handle kernel paging request at virtual address 00000000, epc == 8f3f0a68, ra == 8f3f0d08
[    9.189148] Oops[#1]:
[    9.191691] CPU: 3 PID: 761 Comm: kmodloader Not tainted 4.4.14 #1
[    9.198574] task: 8f18d758 ti: 8f308000 task.ti: 8f308000
[    9.204582] $ 0   : 00000000 00000001 00000000 00000000
[    9.210423] $ 4   : 000000f0 00000001 00000002 00000000
[    9.216262] $ 8   : 00000002 00010000 ffffffff 6566736e
[    9.222102] $12   : 00000000 00000040 00000000 73656c72
[    9.227943] $16   : 8f1f8bc0 00000000 8f1f8c50 8f08d400
[    9.233782] $20   : 00000000 8f1f8ce4 8f1f8d3c 8f3f34d4
[    9.239622] $24   : 00000002 84351614                  
[    9.245461] $28   : 8f308000 8f3098e0 00000006 8f3f0d08
[    9.251302] Hi    : 00000001
[    9.254506] Lo    : 00000000
[    9.257755] epc   : 8f3f0a68 init_module+0x314a68/0x316e40 [ca8210sm]
[    9.264949] ra    : 8f3f0d08 init_module+0x314d08/0x316e40 [ca8210sm]
[    9.272117] Status: 1100fc03 KERNEL EXL IE 
[    9.276794] Cause : 00800008 (ExcCode 02)
[    9.281260] BadVA : 00000000
[    9.284471] PrId  : 0001a120 (MIPS interAptiv (multi))
[    9.290188] Modules linked in: ca8210sm(+) mac802154 ieee802154_6lowpan ieee802154_socket ieee802154 6lowpan snd_soc_core i2c_dev ledtrig_heartbeat ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_commono
[    9.329577] Process kmodloader (pid: 761, threadinfo=8f308000, task=8f18d758, tls=77813e50)
[... 28 lines cut ...]
[   12.332487] [<8e9976a4>] hal_stop+0x15f0/0x1bb8 [uccp420wlan]
[   12.339039] 
[   12.340750] Mem-Info:
[   12.343319] active_anon:155 inactive_anon:1 isolated_anon:0
[   12.343319]  active_file:432 inactive_file:1451 isolated_file:0
[   12.343319]  unevictable:0 dirty:10 writeback:0 unstable:0
[   12.343319]  slab_reclaimable:416 slab_unreclaimable:1344
[   12.343319]  mapped:349 shmem:4 pagetables:18 bounce:0
[   12.343319]  free:41383 free_pcp:296 free_cma:0
[   12.379043] DMA free:0kB min:0kB low:0kB high:0kB active_anon:0kB inactive_anon:0kB active_file:0kB inactive_file:0kB unevictable:0kB isolated(anon):0kB isolated(file):0kB present:65536kB managed:0kB s
[   12.423377] lowmem_reserve[]: 0 180 180
[   12.427716] Normal free:165532kB min:1720kB low:2148kB high:2580kB active_anon:620kB inactive_anon:4kB active_file:1728kB inactive_file:5804kB unevictable:0kB isolated(anon):0kB isolated(file):0kB preo
[   12.476778] lowmem_reserve[]: 0 0 0
[... cut ...]
[   12.491590] Normal: 7*4kB (UME) 18*8kB (UE) 9*16kB (UM) 6*32kB (U) 5*64kB (UM) 3*128kB (UM) 2*256kB (UE) 0*512kB 0*1024kB 2*2048kB (UE) 39*4096kB (M) = 165564kB
[   12.507723] 1915 total pagecache pages
[   12.511940] 0 pages in swap cache
[   12.515635] Swap cache stats: add 0, delete 0, find 0/0
[   12.521483] Free swap  = 0kB
[   12.524691] Total swap = 0kB
[   12.527897] 65536 pages RAM
[   12.531038] 0 pages HighMem/MovableOnly
[   12.535309] 19237 pages reserved
[   12.538903] UCCP420_WIFI_HAL: uccp host ram: failed to allocate memory
[   12.546355] uccp420: probe of 18480000.uccp failed with error -12

francois-berder-imgtec, 2017-01-10:
It was found while working on task Creatordev/openwrt#132 that this patch can be applied if  the kernel is loaded at address 0x80400000.

francois-berder-imgtec, 2017-01-10:
This is blocked until CreatorDev/openwrt#132 is done.

Ham22, 2017-01-10:
If all working can you also make sure that https://github.com/CreatorDev/linux/ is all in sync.

closed 2017-01-12 by mtusnio