CreatorDev/openwrt #85: Trial release tag

Shpinkso, 2016-11-21:
Test that the process for tagging a build works with the CI engine.
i.e.
- Tags repos
- Produces and retains all required artiacts (including opkg packages)

nikhil-zinjurde-imgtec, 2016-11-28:
http://jenkins.creatordev.io/job/CreatorDev/job/openwrt/job/ci40/81/ started to trial a tagged release.

Build options ticked:
* BUILD_TOOLS
* BUILD_ALL_PACKAGES
* VERSION = v0.10.5

nikhil-zinjurde-imgtec, 2016-11-29:
http://jenkins.creatordev.io/job/CreatorDev/job/openwrt/job/ci40/92/ started as a new build with following Build options:

* BUILD_TOOLS enabled
* BUILD_ALL_PACKAGES enabled
* VERSION = v0.10.5
* OVERRIDE_PACKAGES = creator
* OVERRIDE_CI40-PLATFORM-FEED = master
* OVERRIDE_CREATOR-FEED = master

nikhil-zinjurde-imgtec, 2016-11-30:
http://jenkins.creatordev.io/job/CreatorDev/job/openwrt/job/ci40/98/ New build created with above build options.

Verified that
* Image contains v0.10.5 version 
* All packages including Ci40 platform and creator feeds are generated
* Image boots with different boot methods and all interfaces are up and working
* OPKG works with packages hosted on a local server

closed 2016-11-30 by nikhil-zinjurde-imgtec