CreatorDev/openwrt #89: production JIG updates

Ham22, 2016-11-22:
At minimum we will need to update the programming JIGs but likely the testing JIGs will need some sort of change too.

Ham22, 2017-01-18:
Blocked on having final images that can boot (including the prodtest variant)

nikhil-zinjurde-imgtec, 2017-02-21:
@Shpinkso @Ham22 this task has little or no relevance to CreatorDev/openwrt, this could be more relevant to be in u-boot repo since the one of final images required for this task to be carried out is the bootloader image, the other image being the buildroot prodtest variant (not openwrt)

Ham22, 2017-02-22:
The tests are in the platform feed, it's nothing to do with bootloader?

Anyhow it all looks to boot nicely. All tests also passed accept for audio loopback so need to investigate this.

Ham22, 2017-02-22:
We should get this running on the post production JIG here so that we can gain some more confidence in the production test updates.

abhijit-mahajani-imgtec, 2017-03-21:
blocked on https://github.com/CreatorDev/embedded-squad-topics/issues/10