CreatorDev/openwrt #126: Interfaces' status is "UNKNOWN"

mtusnio, 2016-12-13:
Running ip link show gives following output on my board:
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state **UNKNOWN** mode DEFAULT group default qlen 1
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc fq_codel state UP mode DEFAULT group default qlen 1000
    link/ether 00:19:f5:89:0b:fa brd ff:ff:ff:ff:ff:ff
3: wpan0: <BROADCAST,NOARP,UP,LOWER_UP> mtu 123 qdisc fq_codel state **UNKNOWN** mode DEFAULT group default qlen 300
    link/ieee802.15.4 00:19:f5:ff:fe:89:0b:fc brd ff:ff:ff:ff:ff:ff:ff:ff
5: tayga-lan: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP> mtu 1500 qdisc fq_codel state UP mode DEFAULT group default qlen 500
    link/none 
6: lowpan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1280 qdisc noqueue state **UNKNOWN** mode DEFAULT group default qlen 1
    link/[825] 00:19:f5:ff:fe:89:0b:fc brd ff:ff:ff:ff:ff:ff:ff:ff

francois-berder-imgtec, 2016-12-15:
It seems that state is UNKNOWN when the interface is up but not connected to an access point.

mtusnio, 2016-12-16:
Eth0 status after a reboot:
eth0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc fq_codel state DOWN mode DEFAULT group default qle0
    link/ether 00:19:f5:89:0b:fa brd ff:ff:ff:ff:ff:ff
Additionally I believe the loopback interface shouldn't report as UNKNOWN as shown above?

mtusnio, 2016-12-22:
In the end this seems to behave as intended (and works the same way on other distributions as well), this is no longer an issue.

mtusnio, 2017-01-03:
root@OpenWrt:~# ip link show
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN mode DEFAULT group default qlen 1
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc fq_codel state **UNKNOWN** mode DEFAULT group default qlen 1000

Newest builds are reporting eth0 as UNKNOWN.

mtusnio, 2017-01-04:
Cycling the interface (ifconfig eth0 down & up) resolves the UNKNOWN state.

nikhil-zinjurde-imgtec, 2017-03-03:
v1.1.0 and CI images now report the eth0 interface status correctly, so this looks like no longer an issue?

```
root@OpenWrt:/# ip link show
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN qlen 1
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
2: eth0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc fq_codel state DOWN qlen 1000
    link/ether 00:19:f5:89:15:7e brd ff:ff:ff:ff:ff:ff
3: wpan0: <BROADCAST,NOARP,UP,LOWER_UP> mtu 123 qdisc fq_codel state UNKNOWN qlen 300
    link/[804] 02:19:f5:ff:fe:89:15:80 brd ff:ff:ff:ff:ff:ff:ff:ff
5: lowpan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1280 qdisc noqueue state UNKNOWN qlen 1
    link/[825] 02:19:f5:ff:fe:89:15:80 brd ff:ff:ff:ff:ff:ff:ff:ff
root@OpenWrt:/# [   50.880256] stmmaceth 18140000.ethernet eth0: Link is Up - 10Mbps/Half - flow control off

root@OpenWrt:/# ip link show
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN qlen 1
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc fq_codel state UP qlen 1000
    link/ether 00:19:f5:89:15:7e brd ff:ff:ff:ff:ff:ff
3: wpan0: <BROADCAST,NOARP,UP,LOWER_UP> mtu 123 qdisc fq_codel state UNKNOWN qlen 300
    link/[804] 02:19:f5:ff:fe:89:15:80 brd ff:ff:ff:ff:ff:ff:ff:ff
5: lowpan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1280 qdisc noqueue state UNKNOWN qlen 1
    link/[825] 02:19:f5:ff:fe:89:15:80 brd ff:ff:ff:ff:ff:ff:ff:ff
```

nikhil-zinjurde-imgtec, 2017-03-09:
Current OpenWrt images are not showing this failure.  Boardfarm test has been updated to capture such failure in future.

closed 2017-03-09 by nikhil-zinjurde-imgtec