CreatorDev/openwrt #127: Bluetooth cannot scan if the wifi interface is not up

tjobanpu, 2016-12-14:
- WiFi  is disabled by default, when ci40 boots up the /etc/init.d/bt-csr  initializes the Bluetooth but the command "hcitool scan" does not return results.

- When the WiFi interface is up the command "hcitool scan" returns results.

- We suspected that it could be related to the WiFi firmware not  loaded when the wifi driver has been initialized, but our testing showed that even if the WiFi firmware is loaded the Bluetooth does not scan

-Only when you do "wifi up" and can see the wifi interface in the "ifconfig", the BT scan works.

- This looks like a WiFi-BT coex issue.

Ham22, 2017-01-05:
Blocked on waiting for answer from Ensigma firmware team

Shpinkso, 2017-01-12:
Blocked on 6.9 WLAN release

tjobanpu, 2017-01-20:
Have tested the latest V6.9 release of WiFi, confirm that the issue has been fixed.

abhijit-mahajani-imgtec, 2017-01-23:
With 6.9 release, this issue gets fixed.

closed 2017-01-23 by abhijit-mahajani-imgtec