CreatorDev/openwrt #128: Cannot build existing applications using ci40-v1.0.0 openwrt sdk

abhijit-mahajani-imgtec, 2016-12-15:
1.Untar the OpenWrt-SDK-ci40-v1.0.0-pistachio-marduk_gcc-5.3.0_musl-1.1.15.Linux-x86_64.tar.bz2
2.Go into sdk directory and do 
./scripts/feeds update -a;./scripts/feeds install -a
3. make package/ca8210/compile V=s

it gives following error:

/media/abhijit/mak/OpenWrt-SDK-ci40-v1.0.0-pistachio-marduk_gcc-5.3.0_musl-1.1.15.Linux-x86_64/staging_dir/toolchain-mipsel_mips32_gcc-5.3.0_musl-1.1.15/bin/../libexec/gcc/mipsel-openwrt-linux-musl/5.3.0/cc1: /usr/lib/x86_64-linux-gnu/libstdc++.so.6: version `GLIBCXX_3.4.21' not found (required by /media/abhijit/mak/OpenWrt-SDK-ci40-v1.0.0-pistachio-marduk_gcc-5.3.0_musl-1.1.15.Linux-x86_64/staging_dir/toolchain-mipsel_mips32_gcc-5.3.0_musl-1.1.15/bin/../libexec/gcc/mipsel-openwrt-linux-musl/5.3.0/cc1)
make[3]: Entering directory `/media/abhijit/mak/OpenWrt-SDK-ci40-v1.0.0-pistachio-marduk_gcc-5.3.0_musl-1.1.15.Linux-x86_64/build_dir/target-mipsel_mips32_musl-1.1.15/linux-pistachio_marduk/linux-4.4.14'
Makefile:653: Cannot use CONFIG_CC_STACKPROTECTOR_REGULAR: -fstack-protector not supported by compiler
  LD      /media/abhijit/mak/OpenWrt-SDK-ci40-v1.0.0-pistachio-marduk_gcc-5.3.0_musl-1.1.15.Linux-x86_64/build_dir/target-mipsel_mips32_musl-1.1.15/ca8210-1.2.0/Module/built-in.o
  CC [M]  /media/abhijit/mak/OpenWrt-SDK-ci40-v1.0.0-pistachio-marduk_gcc-5.3.0_musl-1.1.15.Linux-x86_64/build_dir/target-mipsel_mips32_musl-1.1.15/ca8210-1.2.0/Module/ca8210.o
/media/abhijit/mak/OpenWrt-SDK-ci40-v1.0.0-pistachio-marduk_gcc-5.3.0_musl-1.1.15.Linux-x86_64/staging_dir/toolchain-mipsel_mips32_gcc-5.3.0_musl-1.1.15/bin/../libexec/gcc/mipsel-openwrt-linux-musl/5.3.0/cc1: /usr/lib/x86_64-linux-gnu/libstdc++.so.6: version `GLIBCXX_3.4.21' not found (required by /media/abhijit/mak/OpenWrt-SDK-ci40-v1.0.0-pistachio-marduk_gcc-5.3.0_musl-1.1.15.Linux-x86_64/staging_dir/toolchain-mipsel_mips32_gcc-5.3.0_musl-1.1.15/bin/../libexec/gcc/mipsel-openwrt-linux-musl/5.3.0/cc1)
as: unrecognized option '-G'
make[4]: *** [/media/abhijit/mak/OpenWrt-SDK-ci40-v1.0.0-pistachio-marduk_gcc-5.3.0_musl-1.1.15.Linux-x86_64/build_dir/target-mipsel_mips32_musl-1.1.15/ca8210-1.2.0/Module/ca8210.o] Error 1
make[3]: *** [_module_/media/abhijit/mak/OpenWrt-SDK-ci40-v1.0.0-pistachio-marduk_gcc-5.3.0_musl-1.1.15.Linux-x86_64/build_dir/target-mipsel_mips32_musl-1.1.15/ca8210-1.2.0/Module] Error 2
make[3]: Leaving directory `/media/abhijit/mak/OpenWrt-SDK-ci40-v1.0.0-pistachio-marduk_gcc-5.3.0_musl-1.1.15.Linux-x86_64/build_dir/target-mipsel_mips32_musl-1.1.15/linux-pistachio_marduk/linux-4.4.14'
make[2]: *** [/media/abhijit/mak/OpenWrt-SDK-ci40-v1.0.0-pistachio-marduk_gcc-5.3.0_musl-1.1.15.Linux-x86_64/build_dir/target-mipsel_mips32_musl-1.1.15/ca8210-1.2.0/.built] Error 2
make[2]: Leaving directory `/media/abhijit/mak/OpenWrt-SDK-ci40-v1.0.0-pistachio-marduk_gcc-5.3.0_musl-1.1.15.Linux-x86_64/feeds/ci40/ca8210'
make[1]: *** [package/feeds/ci40/ca8210/compile] Error 2
make[1]: Leaving directory `/media/abhijit/mak/OpenWrt-SDK-ci40-v1.0.0-pistachio-marduk_gcc-5.3.0_musl-1.1.15.Linux-x86_64'
make: *** [package/ca8210/compile] Error 2

Avinash95, 2016-12-20:
Following commands fixed the issue for me
sudo add-apt-repository ppa:ubuntu-toolchain-r/test
sudo apt-get update
sudo apt-get install gcc-4.9
sudo apt-get upgrade libstdc++6

Ham22, 2016-12-21:
I've not been able to reproduce this (using docker env) and it seems those who can are able to fix it by updating env so closing this?

closed 2016-12-21 by Ham22