CreatorDev/openwrt #145: Opkg update fails to update the package list

mtusnio, 2016-12-20:
Building an image from the tip and running opkg update returns following output:

root@OpenWrt:~# opkg update
Downloading https://downloads.creatordev.io/pistachio/marduk/packages/base/Packages.gz.
wget: SSL support not available, please install ustream-ssl
Downloading https://downloads.creatordev.io/pistachio/marduk/packages/kernel/Packages.gz.
wget: SSL support not available, please install ustream-ssl
Downloading https://downloads.creatordev.io/pistachio/marduk/packages/ci40/Packages.gz.
wget: SSL support not available, please install ustream-ssl
Downloading https://downloads.creatordev.io/pistachio/marduk/packages/ckt/Packages.gz.
wget: SSL support not available, please install ustream-ssl
Downloading https://downloads.creatordev.io/pistachio/marduk/packages/creatordev/Packages.gz.
wget: SSL support not available, please install ustream-ssl
Downloading https://downloads.creatordev.io/pistachio/marduk/packages/luci/Packages.gz.
wget: SSL support not available, please install ustream-ssl
Downloading https://downloads.creatordev.io/pistachio/marduk/packages/management/Packages.gz.
wget: SSL support not available, please install ustream-ssl
Downloading https://downloads.creatordev.io/pistachio/marduk/packages/packages/Packages.gz.
wget: SSL support not available, please install ustream-ssl
Downloading https://downloads.creatordev.io/pistachio/marduk/packages/routing/Packages.gz.
wget: SSL support not available, please install ustream-ssl
Downloading https://downloads.creatordev.io/pistachio/marduk/packages/telephony/Packages.gz.
wget: SSL support not available, please install ustream-ssl
Collected errors:
 * opkg_conf_set_option: Unrecognized option: signature_ca_file=/etc/ssl/certs/opkg.pem
 * opkg_download: Failed to download https://downloads.creatordev.io/pistachio/marduk/packages/base/Packages.gz, wget returned 1.
 * opkg_download: Failed to download https://downloads.creatordev.io/pistachio/marduk/packages/kernel/Packages.gz, wget returned 1.
 * opkg_download: Failed to download https://downloads.creatordev.io/pistachio/marduk/packages/ci40/Packages.gz, wget returned 1.
 * opkg_download: Failed to download https://downloads.creatordev.io/pistachio/marduk/packages/ckt/Packages.gz, wget returned 1.
 * opkg_download: Failed to download https://downloads.creatordev.io/pistachio/marduk/packages/creatordev/Packages.gz, wget returned 1.
 * opkg_download: Failed to download https://downloads.creatordev.io/pistachio/marduk/packages/luci/Packages.gz, wget returned 1.
 * opkg_download: Failed to download https://downloads.creatordev.io/pistachio/marduk/packages/management/Packages.gz, wget returned 1.
 * opkg_download: Failed to download https://downloads.creatordev.io/pistachio/marduk/packages/packages/Packages.gz, wget returned 1.
 * opkg_download: Failed to download https://downloads.creatordev.io/pistachio/marduk/packages/routing/Packages.gz, wget returned 1.
 * opkg_download: Failed to download https://downloads.creatordev.io/pistachio/marduk/packages/telephony/Packages.gz, wget returned 1.

nikhil-zinjurde-imgtec, 2017-01-02:
@mtusnio, I built an image from the tip (ci40 branch of CreatorDev/openwrt, commit  atf6a5f05), ran opkg update, it worked fine for me. Are you sure that you built the image from ci40 branch? The config file in target/linux/pistachio/creator-platform-default-cascoda.config does not use "https", somehow you have that in your image, which is not needed, and which is what is causing these errors.  You can work with simply "http" location of downloads server

mtusnio, 2017-01-03:
@nikhil-zinjurde-imgtec The issue is resolved, I did a distclean build and it switched the https addresses to http, so it's all good now. :)

closed 2017-01-03 by mtusnio