CreatorDev/openwrt #154: sysupgrade does not work

nikhil-zinjurde-imgtec, 2017-01-03:
Doing sysupgrade using the openwrt-x.x.x-pistachio-marduk-marduk_ca8210-squashfs-sysupgrade.tar file, it fails with "Invalid sysupgrade file, Image check 'platform_check_image' failed" error.

Steps followed as mentioned in https://docs.creatordev.io/ci40/guides/transitional-guide/#upgrading-your-ci40-software

nikhil-zinjurde-imgtec, 2017-01-03:
sysupgrade works with openwrt-x.x.x-pistachio-marduk-marduk_ca8210-squashfs-factory.ubi, the documentation on https://docs.creatordev.io/ci40/guides/transitional-guide/#upgrading-your-ci40-software needs to mention the correct file to be used for sysupgrade

Ham22, 2017-01-04:
I think this is due to the device definition in target/linux/pistachio/image/Makfile (marduk_ca8210) not matching the board name (Marduk) which is used for the check but needs to be checked.

abhijit-mahajani-imgtec, 2017-01-16:
Waiting from upstream regarding the suggestion whether to use DEVICES approach or existing PROFILE approach for different h/w boards

Ham22, 2017-01-18:
Should this be closed as duplicate as it's now a subset of https://github.com/CreatorDev/openwrt/issues/216 ?

abhijit-mahajani-imgtec, 2017-02-03:
After #279, it gives different error:

```
root@OpenWrt:/tmp# sysupgrade -n /tmp/openwrt-j9-pistachio-marduk-squashfs-sysup
grade.tar
Sending TERM to remaining processes ... logd rpcd netifd odhcpd tcsd uhttpd dbus-daemon bluetoothd hciattach ntpd dnsmasq ubu
Sending KILL to remaining processes ...
Switching to ramdisk...
[   63.068402] UBIFS (ubi0:2): background thread "ubifs_bgt0_2" stops
Performing system upgrade...
Current boot partiton  1
Writing image to  firmware0
ubidetach: error!: cannot detach "/dev/mtd4"
           error 19 (No such device)
ubiformat: mtd4 (nand), size 268435456 bytes (256.0 MiB), 1024 eraseblocks of 262144 bytes (256.0 KiB), min. I/O size 4096 by
libscan: scanning erase[   68.274669] ubi1: attaching mtd4
libscan: scanning eraseblock 1023 -- 100 % complete
ubiformat: 1024 eraseblocks have valid erase counter, mean value is 61
ubiformat: error!: file "/tmp/openwrt-j9-pistachio-marduk-squashfs-sysupgrade.tar" (size 11642880 bytes) is not multiple of e
           error 0 (No error information)

```

Shpinkso, 2017-02-13:
@abhijit-mahajani-imgtec is this ticket closed now?

Ham22, 2017-02-15:
When working on this for lede port I found that the upgrade scripts for marduk are hardcoded to only work on ubi files. I've fixed it for lede. This ticket only needs to stay open if you wish to backport it.

closed 2017-03-06 by Ham22