CreatorDev/openwrt #171: Add weekly full build to Jenkins

Shpinkso, 2017-01-05:
build should be the full SDK build as we do with releases.
Starts at 00:00 Saturday
Notifies embedded team of failure by email

nikhil-zinjurde-imgtec, 2017-01-11:
The objective of this task cannot be achieved in a straightforward way.  A scheduled build job can be easily added in Jenkinsfile.  However, there is a limitation that there is no way to pass parameters.

The specified default values will indeed be used for string, boolean and choice parameters.  But, since we want the full Build Tools (SDK, ImageBuilder, Toolchain) and all the packages to be built just as we do for releases, this cannot be done as the default value for ALL_PACKAGES and BUILD_TOOLS parameter is "false".  Reference on this limitation: Limitations section in https://wiki.jenkins-ci.org/display/JENKINS/Parameterized+Build

Also, it is currently not possible to build periodically with parameters in pipeline jobs, where we can specify the values to override the default ones. Reference to this issue: https://issues.jenkins-ci.org/browse/JENKINS-37139 and Open Issues section in https://wiki.jenkins-ci.org/display/JENKINS/Parameterized+Build

Ham22, 2017-01-11:
Another open ticket https://issues.jenkins-ci.org/browse/JENKINS-16352 that looks related, the comments suggest a solution soon but we all know what 5mins means in SW time :D

nikhil-zinjurde-imgtec, 2017-01-12:
thanks @Ham22, as discussed, we can then simply run a cron job from Jenkins build master to run following command on 12:00AM, every Saturday

    0 0 * * 6 CRUMB=$(curl -s 'http://USER:PASSWORD@jenkins.creatordev.io/crumbIssuer/api/xml?xpath=concat(//crumbRequestField,":",//crumb)') && curl -v -H $CRUMB -X POST 'http://USER:PASSWORD@jenkins.creatordev.io/job/CreatorDev/job/openwrt/job/ci40/buildWithParameters?BUILD_TOOLS=true&ALL_PACKAGES=true'

nikhil-zinjurde-imgtec, 2017-01-12:
Blocked on Jenkins support

Shpinkso, 2017-01-20:
@abhijit-mahajani-imgtec @nikhil-zinjurde-imgtec the full builds are happening right? Presumably we close this and add a new ticket for proper scheduling, rather than the cron hack (plus the e-mailing task)

nikhil-zinjurde-imgtec, 2017-01-23:
@Shpinkso yes, last weekly build v1.0.2 triggered and completely successfully http://jenkins.creatordev.io/job/CreatorDev/job/openwrt/job/ci40/249/artifact/bin/pistachio/.

However, this ticket was kept open because it's acceptance criteria is in fact proper scheduling and mailing list