CreatorDev/openwrt #172: Wifi displays 'network unreachable' after connecting/disconnecting ethernet

payalsarnot, 2017-01-05:
I have a ci40 board running openwrt 0.10.4 and connected to wifi. Now if I ping network now it works but, after connecting and disconnecting ethernet ping over wifi fails displaying 'Network Unreachable'.

Shpinkso, 2017-01-06:
@payalsarnot please confirm and let me know whether to close the issue or not.

payalsarnot, 2017-01-09:
@Shpinkso @nikhil-zinjurde-imgtec this issue is still present on the latest branch code.

nikhil-zinjurde-imgtec, 2017-01-10:
Confirmed that this is still an issue on the latest ci40 branch. 

Steps to reproduce:

1. Routing table when both wifi and ethernet are connected

        root@OpenWrt:/# route
        Kernel IP routing table
        Destination     Gateway         Genmask         Flags Metric Ref    Use Iface
        default         10.2.2.3        0.0.0.0         UG    0      0        0 wlan0
        10.2.2.0        *               255.255.255.0   U     0      0        0 wlan0
        10.2.2.3        *               255.255.255.255 UH    0      0        0 wlan0
        192.168.91.0    *               255.255.255.0   U     0      0        0 eth0
        192.168.91.254  *               255.255.255.255 UH    0      0        0 eth0

        root@OpenWrt:/# ping google.com
        PING google.com (216.58.201.14): 56 data bytes
        64 bytes from 216.58.201.14: seq=1 ttl=42 time=306.195 ms
        64 bytes from 216.58.201.14: seq=2 ttl=42 time=383.437 ms
        64 bytes from 216.58.201.14: seq=3 ttl=42 time=408.668 ms

2. Routing table when ethernet is disconnected

        root@OpenWrt:/# [  644.261375] stmmaceth 18140000.ethernet eth0: Link is Down

        root@OpenWrt:/# route
        Kernel IP routing table
        Destination     Gateway         Genmask         Flags Metric Ref    Use Iface
        default         10.2.2.3        0.0.0.0         UG    0      0        0 wlan0
        10.2.2.0        *               255.255.255.0   U     0      0        0 wlan0
        10.2.2.3        *               255.255.255.255 UH    0      0        0 wlan0

        root@OpenWrt:/# ping google.com
        PING google.com (216.58.201.14): 56 data bytes
        64 bytes from 216.58.201.14: seq=0 ttl=42 time=398.698 ms
        64 bytes from 216.58.201.14: seq=1 ttl=42 time=405.318 ms
        64 bytes from 216.58.201.14: seq=2 ttl=42 time=686.499 ms

3. Routing table when ethernet is connected back again

        root@OpenWrt:/# [  784.275939] stmmaceth 18140000.ethernet eth0: Link is Up - 10Mbps/Half - flow control off

        root@OpenWrt:/# route
        Kernel IP routing table
        Destination     Gateway         Genmask         Flags Metric Ref    Use Iface
        default         192.168.91.254  0.0.0.0         UG    0      0        0 eth0
        10.2.2.0        *               255.255.255.0   U     0      0        0 wlan0
        10.2.2.3        *               255.255.255.255 UH    0      0        0 wlan0
        192.168.91.0    *               255.255.255.0   U     0      0        0 eth0
        192.168.91.254  *               255.255.255.255 UH    0      0        0 eth0

        root@OpenWrt:/# ping google.com
        PING google.com (216.58.197.46): 56 data bytes
        64 bytes from 216.58.197.46: seq=0 ttl=55 time=39.108 ms
        64 bytes from 216.58.197.46: seq=1 ttl=55 time=29.822 ms
        64 bytes from 216.58.197.46: seq=2 ttl=55 time=30.493 ms

4. Finally, routing table when ethernet is disconnected again

        root@OpenWrt:/# [  871.292106] stmmaceth 18140000.ethernet eth0: Link is Down

        root@OpenWrt:/# route
        Kernel IP routing table
        Destination     Gateway         Genmask         Flags Metric Ref    Use Iface
        10.2.2.0        *               255.255.255.0   U     0      0        0 wlan0
        10.2.2.3        *               255.255.255.255 UH    0      0        0 wlan0

        root@OpenWrt:/# ping google.com
        PING google.com (216.58.197.46): 56 data bytes
        ping: sendto: Network unreachable

nikhil-zinjurde-imgtec, 2017-01-10:
The issue is not fixed either by
1. Commenting or deleting the line "option 'defaultroute' '0'" from /etc/config/network file or even from /etc/uci-defaults/config/network file
OR
2. Changing the defaultroute to 1 in the line "option 'defaultroute' '0'" in /etc/config/network file and/or in /etc/uci-defaults/config/network file

The issue can however be fixed by restarting network using "/etc/init.d/network restart" command

tjobanpu, 2017-01-18:
This is a known issue in the openwrt, 

https://wiki.openwrt.org/doc/uci/network

 It seems that if an interface is configured as dhcp client, at least on OpenWrt 10.03, the default route received by dhcp will be the only one listed and will remove other default route/metrics defined for other interfaces if those interfaces comes "before" the interface with dhcp in terms of "ifname" values.

The symtom is we would not be able to access internet if the ethernet is plugged in and then out, since both the wifi interface and the ethernet interface requires DHCP the above solution is not applicable.

Parking this for now as we are going to move to LEDE from openwrt

closed 2017-01-19 by tjobanpu