CreatorDev/openwrt #187: Decouple creator extras from the platform builds

Ham22, 2017-01-10:
Currently the builds, ci and release of the ci40 platform support and the creator extras (provided by creator feed) are all intertwined causing slow release cycles and blockages between un-related components.

Therefore uncouple them so that the platform support builds only worry about exactly that. Then the creator extras can be added using a mix of the SDK and image builder.

This will the two distinct projects/teams to release at their own pace without worrying about each other (as much).

Ham22, 2017-01-10:
Creatorless sdk build can be found here http://jenkins.creatordev.io/job/CreatorDev/job/openwrt/job/PR-188/3/

Ham22, 2017-01-19:
All issues we had when building manually fixed. To complete the task we now just need to automate building Creator-feed from the SDK in the it's Jenkinsfile.

nikhil-zinjurde-imgtec, 2017-01-26:
Creator SDK failing to build Creator feed packages on [CreatorDev >> Creator-feed >> PR31 >> 3](http://jenkins.creatordev.io/job/CreatorDev/job/Creator-feed/view/Pull%20Requests/job/PR-31/3/) due to following error:

    Makefile:758: recipe for target 'Modules/getbuildinfo.o' failed
    make[4]: *** [Modules/getbuildinfo.o] Error 1
    make[4]: Leaving directory '/home/ubuntu/workspace/eatorDev_Creator-feed_PR-31-CH44VMEDIYHPHEOCHPAOID4G622PMHP2GMIP5XKSS3J76ASUE4VQ/creator-sdk/build_dir/target-mipsel_mips32r2_musl-1.1.15/host/Python-3.5.2'
    Makefile:236: recipe for target '/home/ubuntu/workspace/eatorDev_Creator-feed_PR-31-CH44VMEDIYHPHEOCHPAOID4G622PMHP2GMIP5XKSS3J76ASUE4VQ/creator-sdk/build_dir/target-mipsel_mips32r2_musl-1.1.15/host/Python-3.5.2/.built' failed
    make[3]: *** [/home/ubuntu/workspace/eatorDev_Creator-feed_PR-31-CH44VMEDIYHPHEOCHPAOID4G622PMHP2GMIP5XKSS3J76ASUE4VQ/creator-sdk/build_dir/target-mipsel_mips32r2_musl-1.1.15/host/Python-3.5.2/.built] Error 2
    make[3]: Leaving directory '/home/ubuntu/workspace/eatorDev_Creator-feed_PR-31-CH44VMEDIYHPHEOCHPAOID4G622PMHP2GMIP5XKSS3J76ASUE4VQ/creator-sdk/feeds/packages/lang/python3'
    package/Makefile:196: recipe for target 'package/feeds/packages/python3/host/compile' failed
    make[2]: *** [package/feeds/packages/python3/host/compile] Error 2
    make[2]: Leaving directory '/home/ubuntu/workspace/eatorDev_Creator-feed_PR-31-CH44VMEDIYHPHEOCHPAOID4G622PMHP2GMIP5XKSS3J76ASUE4VQ/creator-sdk'
    package/Makefile:193: recipe for target '/home/ubuntu/workspace/eatorDev_Creator-feed_PR-31-CH44VMEDIYHPHEOCHPAOID4G622PMHP2GMIP5XKSS3J76ASUE4VQ/creator-sdk/staging_dir/target-mipsel_mips32r2_musl-1.1.15/stamp/.package_compile' failed
    make[1]: *** [/home/ubuntu/workspace/eatorDev_Creator-feed_PR-31-CH44VMEDIYHPHEOCHPAOID4G622PMHP2GMIP5XKSS3J76ASUE4VQ/creator-sdk/staging_dir/target-mipsel_mips32r2_musl-1.1.15/stamp/.package_compile] Error 2
    make[1]: Leaving directory '/home/ubuntu/workspace/eatorDev_Creator-feed_PR-31-CH44VMEDIYHPHEOCHPAOID4G622PMHP2GMIP5XKSS3J76ASUE4VQ/creator-sdk'
    /home/ubuntu/workspace/eatorDev_Creator-feed_PR-31-CH44VMEDIYHPHEOCHPAOID4G622PMHP2GMIP5XKSS3J76ASUE4VQ/creator-sdk/include/toplevel.mk:185: recipe for target 'world' failed
    make: *** [world] Error 2

nikhil-zinjurde-imgtec, 2017-01-26:
The Creator SDK v1 (as well as platform SDK v1.0.1) is missing [e973475](https://github.com/CreatorDev/openwrt/commit/e973475ee378c9cf57ded93600ff5c7eb64a9565) fix for Python3 compilation failure which was merged after v1.0.1 release.

nikhil-zinjurde-imgtec, 2017-01-27:
For evaluation,
* I started a build [CreatorDev >> openwrt >> #266](http://jenkins.creatordev.io/job/CreatorDev/job/openwrt/job/ci40/266/artifact/bin/pistachio/) with BUILD_TOOLS and ALL_PACKAGES enabled on Jenkins
* Used the [OpenWrt-SDK](http://jenkins.creatordev.io/job/CreatorDev/job/openwrt/job/ci40/266/artifact/bin/pistachio/OpenWrt-SDK-v1.0.1-pistachio-marduk_gcc-5.3.0_musl-1.1.15.Linux-x86_64.tar.bz2) to compile the Creator feed packages with necessary mods done

This now results in a further new error below:

    for page in $PAGES; do \
	    sed -e "s%LDVERSION%2.4.44%" \
	            -e 's%ETCDIR%/etc/openldap%g' \
		    -e 's%LOCALSTATEDIR%/var%' \
		    -e 's%SYSCONFDIR%/etc/openldap%' \
		    -e 's%DATADIR%/usr/share/openldap%' \
		    -e 's%SBINDIR%/usr/sbin%' \
		    -e 's%BINDIR%/usr/bin%' \
		    -e 's%LIBDIR%/usr/lib%' \
		    -e 's%LIBEXECDIR%/usr/lib%' \
		    -e 's%MODULEDIR%/usr/lib/openldap%' \
		    -e 's%RELEASEDATE%2016/02/05%' \
			    ./$page \
		    | (cd .; soelim -) > $page.tmp; \
    done
    /bin/sh: 15: soelim: not found
    /bin/sh: 15: soelim: not found
    /bin/sh: 15: soelim: not found
    /bin/sh: 15: soelim: not found
    /bin/sh: 15: soelim: not found
    /bin/sh: 15: soelim: not found
    /bin/sh: 15: soelim: not found
    /bin/sh: 15: soelim: not found
    /bin/sh: 15: soelim: not found
    Makefile:292: recipe for target 'all-common' failed
    make[7]: *** [all-common] Error 127
    make[7]: Leaving directory '/home/build/PR-31/creator-sdk/build_dir/target-mipsel_mips32r2_musl-1.1.15/openldap-2.4.44/doc/man/man1'
    Makefile:290: recipe for target 'all-common' failed
    make[6]: *** [all-common] Error 1
    make[6]: Leaving directory '/home/build/PR-31/creator-sdk/build_dir/target-mipsel_mips32r2_musl-1.1.15/openldap-2.4.44/doc/man'
    Makefile:290: recipe for target 'all-common' failed
    make[5]: *** [all-common] Error 1
    make[5]: Leaving directory '/home/build/PR-31/creator-sdk/build_dir/target-mipsel_mips32r2_musl-1.1.15/openldap-2.4.44/doc'
    Makefile:1348: recipe for target 'all-common' failed
    make[4]: *** [all-common] Error 1
    make[4]: Leaving directory '/home/build/PR-31/creator-sdk/build_dir/target-mipsel_mips32r2_musl-1.1.15/openldap-2.4.44'
    Makefile:149: recipe for target '/home/build/PR-31/creator-sdk/build_dir/target-mipsel_mips32r2_musl-1.1.15/openldap-2.4.44/.built' failed
    make[3]: *** [/home/build/PR-31/creator-sdk/build_dir/target-mipsel_mips32r2_musl-1.1.15/openldap-2.4.44/.built] Error 2
    make[3]: Leaving directory '/home/build/PR-31/creator-sdk/feeds/packages/libs/openldap'
    package/Makefile:196: recipe for target 'package/feeds/packages/openldap/compile' failed
    make[2]: *** [package/feeds/packages/openldap/compile] Error 2
    make[2]: Leaving directory '/home/build/PR-31/creator-sdk'
    package/Makefile:193: recipe for target '/home/build/PR-31/creator-sdk/staging_dir/target-mipsel_mips32r2_musl-1.1.15/stamp/.package_compile' failed
    make[1]: *** [/home/build/PR-31/creator-sdk/staging_dir/target-mipsel_mips32r2_musl-1.1.15/stamp/.package_compile] Error 2
    make[1]: Leaving directory '/home/build/PR-31/creator-sdk'
    /home/build/PR-31/creator-sdk/include/toplevel.mk:186: recipe for target 'world' failed
    make: *** [world] Error 2

closed 2017-01-30 by nikhil-zinjurde-imgtec