CreatorDev/openwrt #198: Get OpenWrt to compile with creatordev/linux kernel

mtusnio, 2017-01-12:
At the moment fails with:

```
In file included from /home/tusnio/ci40_project/openwrt/staging_dir/toolchain-mipsel_mips32r2_gcc-5.3.0_musl-1.1.15/include/linux/if_bridge.h:18:0,
                 from /home/tusnio/ci40_project/openwrt/build_dir/target-mipsel_mips32r2_musl-1.1.15/netifd-2016-06-06/system-linux.c:36:
/home/tusnio/ci40_project/openwrt/staging_dir/toolchain-mipsel_mips32r2_gcc-5.3.0_musl-1.1.15/include/linux/in6.h:32:8: error: redefinition of 'struct in6_addr'
 struct in6_addr {
        ^
In file included from /home/tusnio/ci40_project/openwrt/staging_dir/toolchain-mipsel_mips32r2_gcc-5.3.0_musl-1.1.15/include/arpa/inet.h:9:0,
                 from /home/tusnio/ci40_project/openwrt/build_dir/target-mipsel_mips32r2_musl-1.1.15/netifd-2016-06-06/system-linux.c:27:
/home/tusnio/ci40_project/openwrt/staging_dir/toolchain-mipsel_mips32r2_gcc-5.3.0_musl-1.1.15/include/netinet/in.h:23:8: note: originally defined here
 struct in6_addr {
        ^
In file included from /home/tusnio/ci40_project/openwrt/staging_dir/toolchain-mipsel_mips32r2_gcc-5.3.0_musl-1.1.15/include/linux/if_bridge.h:18:0,
                 from /home/tusnio/ci40_project/openwrt/build_dir/target-mipsel_mips32r2_musl-1.1.15/netifd-2016-06-06/system-linux.c:36:
/home/tusnio/ci40_project/openwrt/staging_dir/toolchain-mipsel_mips32r2_gcc-5.3.0_musl-1.1.15/include/linux/in6.h:49:8: error: redefinition of 'struct sockaddr_in6'
 struct sockaddr_in6 {
        ^
In file included from /home/tusnio/ci40_project/openwrt/staging_dir/toolchain-mipsel_mips32r2_gcc-5.3.0_musl-1.1.15/include/arpa/inet.h:9:0,
                 from /home/tusnio/ci40_project/openwrt/build_dir/target-mipsel_mips32r2_musl-1.1.15/netifd-2016-06-06/system-linux.c:27:
/home/tusnio/ci40_project/openwrt/staging_dir/toolchain-mipsel_mips32r2_gcc-5.3.0_musl-1.1.15/include/netinet/in.h:34:8: note: originally defined here
 struct sockaddr_in6 {
        ^
In file included from /home/tusnio/ci40_project/openwrt/staging_dir/toolchain-mipsel_mips32r2_gcc-5.3.0_musl-1.1.15/include/linux/if_bridge.h:18:0,
                 from /home/tusnio/ci40_project/openwrt/build_dir/target-mipsel_mips32r2_musl-1.1.15/netifd-2016-06-06/system-linux.c:36:
/home/tusnio/ci40_project/openwrt/staging_dir/toolchain-mipsel_mips32r2_gcc-5.3.0_musl-1.1.15/include/linux/in6.h:59:8: error: redefinition of 'struct ipv6_mreq'
 struct ipv6_mreq {
        ^
In file included from /home/tusnio/ci40_project/openwrt/staging_dir/toolchain-mipsel_mips32r2_gcc-5.3.0_musl-1.1.15/include/arpa/inet.h:9:0,
                 from /home/tusnio/ci40_project/openwrt/build_dir/target-mipsel_mips32r2_musl-1.1.15/netifd-2016-06-06/system-linux.c:27:
/home/tusnio/ci40_project/openwrt/staging_dir/toolchain-mipsel_mips32r2_gcc-5.3.0_musl-1.1.15/include/netinet/in.h:42:8: note: originally defined here
 struct ipv6_mreq {
        ^
make[6]: *** [CMakeFiles/netifd.dir/system-linux.c.o] Error 1
make[6]: Leaving directory `/home/tusnio/ci40_project/openwrt/build_dir/target-mipsel_mips32r2_musl-1.1.15/netifd-2016-06-06'
make[5]: *** [CMakeFiles/netifd.dir/all] Error 2
make[5]: Leaving directory `/home/tusnio/ci40_project/openwrt/build_dir/target-mipsel_mips32r2_musl-1.1.15/netifd-2016-06-06'
make[4]: *** [all] Error 2
make[4]: Leaving directory `/home/tusnio/ci40_project/openwrt/build_dir/target-mipsel_mips32r2_musl-1.1.15/netifd-2016-06-06'
make[3]: *** [/home/tusnio/ci40_project/openwrt/build_dir/target-mipsel_mips32r2_musl-1.1.15/netifd-2016-06-06/.built] Error 2
make[3]: Leaving directory `/home/tusnio/ci40_project/openwrt/package/network/config/netifd'
make[2]: *** [package/network/config/netifd/compile] Error 2
make[2]: Leaving directory `/home/tusnio/ci40_project/openwrt'
make[1]: *** [/home/tusnio/ci40_project/openwrt/staging_dir/target-mipsel_mips32r2_musl-1.1.15/stamp/.package_compile] Error 2
make[1]: Leaving directory `/home/tusnio/ci40_project/openwrt'
```

abhijit-mahajani-imgtec, 2017-01-20:
Depends upon #219

closed 2017-03-06 by abhijit-mahajani-imgtec

Shpinkso, 2017-03-06:
duplicate of linux/#15