CreatorDev/openwrt #199: Test lowpan throughput on new cascoda devices

Shpinkso, 2017-01-13:
Ci40<->Ci40 is a minimum
Ci40<->Clicker would be nice if there's any tools for this

should be 5 modified Clickers and 5 CI40s in the KL office

Shpinkso, 2017-01-16:
Blocked on only having 1 ci40 with new device fitted, expecting more tomorrow (17.01)

francois-berder-imgtec, 2017-01-19:
It was found that the cascoda driver is not very stable when trying to find the maximal bandwidth.
To reproduce this issue, run ```iperf3 -s -V``` on server and ```iperf3 -c <server-lowpan-IP> -u -t 60 -l 50 -b 0```

At some point, the cascoda driver on the server prints on the shell:
```
[ 2953.700455] ca8210 spi0.0: ca8210 was busy during attempted write
[ 2953.720333] ca8210 spi0.0: spi write retry 1...
```
I had to reboot my board to get 6lowpan working again.

closed 2017-01-19 by francois-berder-imgtec

Ham22, 2017-01-19:
@francois-berder-imgtec has a ticket been created recording the stability issues you found?

francois-berder-imgtec, 2017-01-19:
@Ham22 I've just created a ticket: https://github.com/CreatorDev/openwrt/issues/222

Shpinkso, 2017-01-19:
@francois-berder-imgtec are there existing results from the previous cascoda chips? I.e. do we know if this is a regression or always existed? If not, let's check