CreatorDev/openwrt #217: sometimes sc16is7xx.ko is not installed on the target

abhijit-mahajani-imgtec, 2017-01-18:
It has been observed couple of times that heartbeat led is not flashing once the Ci40 is booted up. Further investigation has showed up that serial driver sc16is7xx.ko is not there in the target file system.
so there is nothing in /sys/class/leds because of which anything related to LEDs does not work. Also sound-pistachio-soc is dependent on this serial-sc16is7xx so audio related tests too fail to work.

In ci40-v1.0.1rc4 image (http://jenkins.creatordev.io/job/CreatorDev/job/openwrt/job/ci40/237/) heartbeat led is not flashing , spi uart tests are failing, audio/mic tests are not working. all these are the end effects of the root cause mentioned above.

Ham22, 2017-01-18:
Does just opkg installing the kmod fix the issue (as a workaround)?

francois-berder-imgtec, 2017-01-18:
@Ham22  Yes it does with this command ``opkg install kmod-serial-sc16is7xx```

francois-berder-imgtec, 2017-01-18:
I followed the instructions found in Jenkinsfile and sc16is7xx.ko is not missing.

Ham22, 2017-02-03:
Running `make kernel_menuconfig` shows the driver as unselected. Looks like CONFIG_SPI and CONFIG_SPI_MASTER were removed from config-4.4 which isn't allowed.

closed 2017-02-20 by abhijit-mahajani-imgtec

Shpinkso, 2017-02-20:
Issue is limited to CI builds, there is a simple manual step that can be taken when needed (Openwrt releases) so we'll take that effort until major releases are switched to LEDE