CreatorDev/openwrt #222: Cascoda driver not stable when testing maximum throughput

francois-berder-imgtec, 2017-01-19:
When doing some thoughput tests between 2 Ci40, it was found that the cascoda driver would always report the error:
```
[ 2953.700455] ca8210 spi0.0: ca8210 was busy during attempted write
[ 2953.720333] ca8210 spi0.0: spi write retry 1...
```

To reproduce this issue, run ```iperf3 -s -V``` on server and ```iperf3 -c <server-lowpan-IP> -u -t 60 -l 50 -b 0```

francois-berder-imgtec, 2017-01-19:
As suggested by Tushar, one thing to try is writing zeros instead of ones in ```spi_write_dummy``` function.

```
diff --git a/Module/ca8210.c b/Module/ca8210.c
index 2aa1565..8113a0c 100644
--- a/Module/ca8210.c
+++ b/Module/ca8210.c
@@ -1167,7 +1167,7 @@ static int ca8210_spi_write_dummy(struct spi_device *spi)
        uint8_t idle = SPI_IDLE;
 
        dev_dbg(&spi->dev, "spi: writing dummy packet\n");
-       ret =  ca8210_spi_write(spi, &idle, 1);
+       ret =  ca8210_spi_write(spi, &idle, 0);
        dev_dbg(&spi->dev, "spi: wrote dummy packet\n");
        return ret;
 }
```

Shpinkso, 2017-01-20:
This is tested on CI40s modified to add new issue Cascoda chips.
This first needs to be compared with current MP CI40s - want to know if it is a regression or not.

Shpinkso, 2017-01-24:
@tjobanpu we're confident this is couldn't be reproduced from Francois test cases?

tjobanpu, 2017-01-25:
yes I had tried a few times (~5) but I was not able to reproduce the issue.

closed 2017-01-25 by Shpinkso