CreatorDev/openwrt #229: Selective CI builds

Shpinkso, 2017-01-20:
As a product owner
I want documentation changes to go straight to review without CI build/tests
so that

nikhil-zinjurde-imgtec, 2017-01-25:
One option could be to ignore commits with certain messages i.e. if the commits include certain words, they can be ignored from triggering a CI build

    checkout([$class: 'GitSCM',
        extensions: scm.extensions + [
            [$class: 'MessageExclusion', excludedMessage: '.*\\[ignore-this-commit\\].*']],
        ],
    ])

Ham22, 2017-01-27:
Comments would allow devs to bypass the CI where as we want the CI to decide right? Filtering the trigger on files paths/names/types seems sensible. Filtering out the triggers may not be the best thing to do however because our process requires a CI pass to allow a merge. Perhaps best would be to just add some logic to the Jenkinsfile to do simpler builds/tests or just return success directly on certain file changes?

nikhil-zinjurde-imgtec, 2017-02-20:
Better approach to take as put here and extend towards different patterns:
https://gist.github.com/lsjostro/3f410333ac90ecccfdc139079f585b3a