CreatorDev/openwrt #232: (Ci40) Flash LEDs on RF traffic

Shpinkso, 2017-01-23:
As a Demonstrator
I want individual LEDs to represent RF traffic across BT, 802.15.4 and wifi
So that I have some visual representation of the three RF components working together

Acceptance:
-Individual LEDs represent each RF component (BT/802.15.4/Wifi)
-LED flashes at RX/TX to show traffic
-LED does not flash if no traffic on corresponding RF component

francois-berder-imgtec, 2017-01-24:
After flashing a usb stick, create a file 01_leds in ```/etc/board.d```:

```
#!/bin/sh

. /lib/functions/uci-defaults.sh

board_config_update
ucidef_set_led_netdev "lan" "lan" "marduk:red:user1" "eth0"
ucidef_set_led_netdev "wifi" "wifi" "marduk:red:user2" "wlan0"
ucidef_set_led_netdev "lowpan" "lowpan" "marduk:red:user3" "lowpan0"
board_config_flush

exit 0
```

nikhil-zinjurde-imgtec, 2017-01-24:
@Shpinkso @francois-berder-imgtec as mentioned in acceptance criteria, BT also supposed to show flashing LED for data (not RF) traffic, so that appears to be missing in the code snippet above

closed 2017-01-30 by francois-berder-imgtec