CreatorDev/openwrt #234: Nodejs fails to run on 1.0.1

mtusnio, 2017-01-23:
Steps to reproduce:

opkg update
opkg install node
node
Returns "Illegal instruction"

mtusnio, 2017-01-26:
Works after enabling softfloat.

Ham22, 2017-01-27:
Is this due to nodejs being built for softfloat so just needs the package to be rebuilt or is node directly calling an api that isn't exposed for hardfloat?

mtusnio, 2017-01-27:
I built node together with my own image of OpenWrt, so it's already built into it.

https://github.com/openwrt/packages/blob/master/lang/node/Makefile#L65 - looking at this it seems there might some issues with the way the if resolves in line 65 & 67

Shpinkso, 2017-02-08:
Obvious question - are we building the latest LTS version of Nodejs?

mtusnio, 2017-02-08:
I don't think we are, but getting that working would be a separate bit. I think OpenWrt packages have some old Node.js running (4.4.5 I believE), but the problem might be in the compilation options.

Ham22, 2017-02-09:
We switched from softfloat to hardfloat but didn't enable the kernel fpu emulator which is needed to kick in whenever some fpu instructions that MIPS don't support are used https://www.linux-mips.org/wiki/Floating_point

closed 2017-02-10 by Ham22