CreatorDev/openwrt #236: Backport all the ASoC related patches from linux upstream to 4.4

abhijit-mahajani-imgtec, 2017-01-23:
Currently we have these many ASoC related patches.  

0011-ASoC-img-Add-Imagination-Technologies-drivers.patch:Subject: ASoC: img: Add Imagination Technologies drivers
0013-ASoC-img-Add-Pistachio-bring-up-board-audio-card-dri.patch:Subject: ASoC: img: Add Pistachio bring-up board audio card driver and its
0023-ASoC-img-Include-audio-pll-in-event_timer.patch:Subject: ASoC: img: Include audio pll in event_timer
0025-ASoC-img-add-rate-change-tolerance.patch:Subject: ASoC: img: add rate change tolerance
0030-ASoC-img-use-non-drifted-rates-when-setting-codec-cl.patch:Subject: ASoC: img: use non-drifted rates when setting codec clocks
0049-ASoC-img-Add-Pistachio-internal-DAC-driver.patch:Subject: ASoC: img: Add Pistachio internal DAC driver
0050-ASoC-img-Card-driver-and-event-timer-rewrite.patch:Subject: ASoC: img: Card driver and event timer rewrite
0051-ASoC-img-rename-pistachio-card-driver-files.patch:Subject: ASoC: img: rename pistachio card driver files
0052-ASoC-img-add-tpa6130a2-hacks.patch:Subject: ASoC: img: add tpa6130a2 hacks
0075-ASoC-img-initialise-Pistachio-active_rate-to-zero.patch:Subject: ASoC: img: initialise Pistachio active_rate to zero
0077-ASoC-img-Fix-64000hz-pll-frequency.patch:Subject: ASoC: img: Fix 64000hz pll frequency
0091-ASoC-img-Fix-compilation-issue-caused-when-selecting.patch:Subject: ASoC: img: Fix compilation issue caused when selecting
0095-ASoC-Update-drivers-for-upstreaming.patch:2 ASoC: img: i2s_out: Upstreaming changes
0095-ASoC-Update-drivers-for-upstreaming.patch:3. ASoC: img: parallel_out: Upstreaming changes
0095-ASoC-Update-drivers-for-upstreaming.patch:4. ASoC: img: spdif_in: Upstreaming changes
0095-ASoC-Update-drivers-for-upstreaming.patch:5. ASoC: img: spdif_out: Upstreaming changes
0095-ASoC-Update-drivers-for-upstreaming.patch:6. ASoC: img: internal_dac: Upstreaming changes
0113-ASoC-img-Add-split-i2s-in.patch:Subject: ASoC: img: Add split i2s-in

and it differs what is there already upstreamed i.e.
3950362 ASoC: img: Add driver for Pistachio internal DAC
7274d07c ASoC: img: Add binding document for Pistachio internal DAC
7f0e823 ASoC: img: parallel out: Add missing initialiser
3958232 ASoC: img: Add driver for SPDIF output controller
0f4ab87 ASoC: img: Add binding document for SPDIF output controller
8f6eec1 ASoC: img: Add documentation for SPDIF in controls
c4458b7 ASoC: img: Add driver for SPDIF input controller
4e8bc8e ASoC: img: Add binding document for SPDIF input controller
8ceb3b2 ASoC: img: Add driver for parallel output controller
71dfaf5 ASoC: img: Add binding document for parallel output controller
d0e3992 ASoC: img: Add driver for I2S output controller
2dcfa06 ASoC: img: Add binding document for I2S output controller
14b947d ASoC: img: Add driver for I2S input controller
e13c118 ASoC: img: Add binding document for I2S input controller

so we need to make sure that patches-4.4 are updated with these backported patches starting from 000-xxx

francois-berder-imgtec, 2017-01-31:
All the upstream patches apply without any conflicts but as mentioned not everything was upstreamed and we will have to re-generate some patches.

francois-berder-imgtec, 2017-02-02:
Already did some work in re-generating patches on top of upstream ones.

closed 2017-02-06 by Ham22