CreatorDev/openwrt #250: Cleanup and Backport clk/pistachio, clocksource/pistachio drivers into 4.4

abhijit-mahajani-imgtec, 2017-01-25:
Currently there are few patches related to clk-pistachio, clocksource, pll
0002-clk-pistachio-Implement-PLL-rate-adjustment.patch
0016-clk-pistachio-Add-cs_RPUCORECLKOUT_1-clock-mux.patch
0024-clk-pistachio-Add-audio-pll-rates-and-set-parent-rat.patch
0065-clk-pistachio-allow-timer-slow-clock-rate-to-be-chan.patch
0072-clk-sdhost-Fix-wrong-SDHost-card-speed.patch
0086-streamsync-Provide-the-atu-clk-adjust-freq.patch

0064-clocksource-pistachio-Fix-configuring-all-timer-cloc.patch
0071-clocksource-pistachio-remove-clock-source-as-valid-i.patch

0070-pistachio-suspend-calibrate-clock-and-pll.patch
0074-Fractional-PLL-freq-adjustments-fine-tuning.patch

This needs to be checked with what is present in upstream, and need to be backported in 4.4, and cleanup the existing patches appropriately. (add new ones/remove what is already there)

More importantly this needs to be matched with the prefix as 
0xx - upstream backports or
1xx - code awaiting upstream merge

This connects to #219

francois-berder-imgtec, 2017-01-25:
0059-pistachio-suspend-Add-support-for-system-suspend.patch introduces arch/mips/pistachio/suspend.S which is modified by 0070-pistachio-suspend-calibrate-clock-and-pll.patch,

francois-berder-imgtec, 2017-01-26:
Removed patch 86 and 74 from the list as it concerns ```atu_clk_maintainer.c```.

francois-berder-imgtec, 2017-01-26:
0079-pistachio-4.4-kernel-migration-fixup.patch needs to be split as it contains changes about ```drivers/clk/pistachio/clk-pll.c```.

closed 2017-02-06 by Ham22