CreatorDev/openwrt #256: BLE connect sometimes does not work

mtusnio, 2017-01-26:
Running this command:
gatttool -b <lightbulb_mac> -I

And then typing "connect" usually gives:
Attempting to connect to D0:B5:C2:B1:A6:77
Error: connect: Connection refused (146)

However, in some cases flashing a brand new image makes it work and BLE has no issues connecting.

tjobanpu, 2017-01-30:
the BLLE connection wont work when you boot from the USB this is because many BT devices use usb as transport and the Bluez software gets confused..
I tried the connection by flashing the software on the board many times
and never failed
D0:B5:C2:B1:A6:77 LEDBlue-C2B1A677                                              
^Croot@OpenWrt:/# gatttool -b D0:B5:C2:B1:A6:77 -I                              
[D0:B5:C2:B1:A6:77][LE]> connect                                                
Attempting to connect to D0:B5:C2:B1:A6:77                                      
Connection successful

Shpinkso, 2017-02-06:
@tjobanpu please update with the latest information

tjobanpu, 2017-02-06:
There was a race condition between the driver and the initialisation scripts, when the driver was enumerating the controller, the bt initialize script pulled the interface down using the "hciconfig hci0 reset" command, the bluetoothd daemon application is not able to find any controller. The solution is to first ensure that the driver has been loaded then initialize the BT controller , then enable the bluetoothd bluez daemon and then configure the interface

closed 2017-02-14 by francois-berder-imgtec