CreatorDev/openwrt #280: Cleanup mtd/spi-nor patches

francois-berder-imgtec, 2017-01-31:
0107-mtd-m25p80-add-support-for-read_xfer-and-write_xfer.patch
0108-mtd-spi-nor-add-support-for-winbond-OTP.patch

Ham22, 2017-02-06:
These patches aren't strictly required for the system to work because the bootloader does the work of reading the OTP.

They cause issue because they conflict with `mtd: devices: m25p80: add support for mmap read request` from the openwrt/lede generic patches meaning generating them from our kernel tree becomes a little bit more complex.

Ham22, 2017-02-28:
Look like we are going to upstream these changes directly to the kernel and seeing as they aren't actually needed right now closing the ticket here.

closed 2017-02-28 by Ham22