CreatorDev/openwrt #293: DEVICE_REVISION in /etc/device_info not 'sensible'

nikhil-zinjurde-imgtec, 2017-02-15:
This is what it is on the latest ci40 branch
    
    root@OpenWrt:/# cat /etc/device_info 
    DEVICE_MANUFACTURER='Imagination Technologies'
    DEVICE_MANUFACTURER_URL='www.imgtec.com'
    DEVICE_PRODUCT='Creator Ci40(marduk)'
    DEVICE_REVISION='v0'

This is what it was in v1.0.1 tag

    root@OpenWrt:/# cat /etc/device_info 
    DEVICE_MANUFACTURER='Imagination Technologies'
    DEVICE_MANUFACTURER_URL='www.imgtec.com'
    DEVICE_PRODUCT='Creator Ci40(Marduk)'
    DEVICE_REVISION='Ci40 with CA8210'

Ham22, 2017-02-15:
I shall kick off an email with marketing and HW teams so they can decide what they want. It's not overly simple as technically there are arguments for a few options:

* blank it out as we only have 1 public version and we don't implement any version detection (this is what we thought we did but seems not setting it defaults to v0)
* v1 (we only ever released one board publicly so far)
* v5+ (this matches the hw id written on the board for all supported boards 5, 7 and 8)
* v7+ (5 never went public and 6 never existed so don't talk about them)

Ham22, 2017-03-06:
Marketing had this response: v7+ preferred, v5+ allowed

PR #299 (2017-03-07): pistachio: set DEVICE_REVISION to v7+
This represents all the h/w versions of Ci40 boards available publicly v7 onwards that will work with the default configuration of marduk platform

The version is not set for marduk_legacy boards as they are no longer in use and hence not supported

This closes #293
closed 2017-03-07 by nikhil-zinjurde-imgtec