CreatorDev/openwrt #304: Web site is closed

ymkyu, 2018-10-22:
I can't access the web site(Creatordev.io). Because it is closed. I want to get docs that guide for CI40.
what can i do for find docs ? And i have a question for CI40.
Connected the UART of CI40 to computer. I can access the Preinstalled that openWRT.
I think u-boot is stored on the nor flash and openwrt is stored on the nand flash. is it right?
I asked to MIPS and Imagination, but they didn't answer to my question.!!!
PLZ, answer my question...

MattAtkinson, 2018-10-30:
Hi ymkyu,

Unfortunately that site was the only location that the docs were stored, so only Imagination/MIPS could possibly get those back (I don't know who's responsible now).

You're correct about the flash.

ymkyu, 2018-10-31:
Thanks for your answer.!